dji-sdk/Onboard-SDK #8: Let's make a standard for md documents

paulyang1990, 2015-06-26:


lanyusea, 2015-06-27:
Thanks for the assignment from Paul.

Since all the comments here will be emailed to all 15 developers watching this repo, I will and I suggest other people post things like a summary/conclusion/issues here instead of general discussions.

---

## The First Draft of onboard SDK Markdown Document Standard

#### The suggested workflow of document update

This part talks about the suggested workflow when updating document

##### Regular Update:

This part describes the regular update when a new official pdf document released.

> pdf official release -> find the diff -> update the markdown file -> review by the merge admin

Every developer is welcome to help us update the markdown file.

##### Non-regular Update:
This part describes the non-regular update whenever an issue is found.

1. Non-regular of conflict.

  When an conflict is found between pdf document and md document, the developer is suggested to report it on issue.

  Any conflict can be merged by any developer through pull request, but he/she **must** refer the merge to the corresponding issue.

  Pull request related to pdf/md conflict *will not* be accepted if no corresponding issue is attached. ie. developers need to create an issue to inform others beforing sending his/her pull request.

2. Non-regular of issue.

  When an issue/bug/mistake is found in our document (pdf and markdown). The developer can create and send a pull request to fix it. 

  He/She **must** refer the pull request with a link to a related discussion, forum.dji.com or bbs.dji.com or github issue or email records, where a SDK staff **must** admit it is a problem in our document.

The pull request of non-regular update does not follow the guildline will not be accepted.

#### The suggested format of Markdown document

This part talks about the document format and syntax. You can refer to the current Markdown document as an example or reference.

1. Lists

  Although the pdf documents use numbered lists, it is suggested to use Hearders instead to separate different sessions.

  Some other numbered lists in pdf file may describing parallel items, where numbered lists or bullet points can be used to express the parallel relation.

2. Tables

  The table in pdf file is more complicated than what markdown syntax supports. It is suggested to use `html` directly instead of markdown table if two or more cells are spanned in table.

  However, to meet the markdown style, the first line of table must be in bold style, the following content should be in normal style.

  i.e.:
  ```html
<table>
<tr>
  <th>Field</th>
  <th>Byte Index</th>
  <th>Size(bit)</th>
  <th>Description</th>
</tr>

<tr>
  <td>SOF</td>
  <td>0</td>
  <td>8</td>
  <td>Frame start byte, fixed to be 0xAA</td>
</tr>
</table>
  ```

3. Images

  All images should be stored in the `Images` folder and developers should refer to the Image using **relative path** instead of absolute path.

  i.e. you need use `![N1UI](Images/N1UI.png)` instead of `![N1UI](https://github.com/dji-sdk/onboard/blob/master/Onboard_API_Doc/Images/N1UI.png)`

4. Codes

   To insert code, we suggest using ``` ``` with code fencing enabled. You can also use `` and indentation but they are not suggested because of lower readability. 

  Example for three different usages:

<pre lang="no-highlight"><code>```c
void get_control_callback(const void* p_data, unsigned int n_size) { }
```

`void get_control_callback(const void* p_data, unsigned int n_size) { }`

      void get_control_callback(const void* p_data, unsigned int n_size) { }
</code></pre>

   Results: 

```c
void get_control_callback(const void* p_data, unsigned int n_size) { }
```

  `void get_control_callback(const void* p_data, unsigned int n_size) { }`

      void get_control_callback(const void* p_data, unsigned int n_size) { }

fritx, 2015-07-10:
so cool!

lanyusea, 2015-09-06:
closed, please ref to the new one #19

closed 2015-09-06 by lanyusea