dji-sdk/Onboard-SDK #22: api_vel_data_t is corrupting the data that follows it.

geetesh, 2015-10-06:
```
typedef struct
{
    fp32 x;
    fp32 y;
    fp32 z;
//    unsigned char health_flag         :1;
//    unsigned char feedback_sensor_id  :4;
//    unsigned char reserve             :3;
}api_vel_data_t;
```
I had to comment out the last three values to fix this issue. I think N1 controller is not outputting this format leading to corruption of all the data that follows it like the global/local position, RC etc.

lanyusea, 2015-10-06:
Hi, did you install Guidance?

geetesh, 2015-10-06:
Yes, I did.

geetesh, 2015-10-11:
Hi, is this issue fixed? I see you have made a new commits but this data structure is till the same. What firmware for N1 controller does this support?

lanyusea, 2015-10-11:
hi @geetesh ,we are still checking if there is something working incorrect in N1 part or in Guidance part.

may I ask the version of your N1 and Guidance firmwares?

geetesh, 2015-10-12:
I think the public firmware for N1 is 2.2.7 and thats why the issue persists. My Guidance sensor is using 1.3.7. When do youplan to release 2.3?

lanyusea, 2015-10-12:
hi @geetech, we have already release 2.3. Please download the new N1
assistant and use it to update your firmware.

https://developer.dji.com/onboard-sdk/downloads/

geetesh, 2015-10-12:
great, will try it and get back to you if the issue persists

lanyusea, 2015-10-12:
just tell me if this issue still resist ;)

lanyusea, 2015-10-14:
Hi @geetesh , does the problem persist? 
if not, I will close this issue.

geetesh, 2015-10-15:
hi sorry for the delay. i'll get back with the results within 24hrs

geetesh, 2015-10-15:
I have been trying to update the FW via N1-Assistant but the dji server seems to be down. I am not able to login or update the firmware. Are you also facing such issues? Im in Pittsburgh, USA

geetesh, 2015-10-16:
Hey, so finally i was able to update to 2.3.2 and the issue is resolved. Thanks :D

closed 2015-10-16 by geetesh

lanyusea, 2015-10-18:
glad to hear that it is solved :)