dji-sdk/Onboard-SDK #40: Onboard SDK serial communication --- NO RESPONSE: 374 ---

nesnes, 2016-02-29:
Hi, I'm trying use the provided samples to gather some informations from the drone and control it in the future. But I'm facing an error both on linux(Ubuntu 14.04) and on windows(7pro) that prevent me to do anything. 
Here is my configuration :
 - I've checked that the remote and the drone software are up to date ;
 - I've generated and activated my api key ;
 - I've wrote those informations in the config file(linux) or in the text-box(windows) ;
 - I've checked the serial speed in the DJI-Assistant, and checked the API activation ;
 - I've carefully selected the good serial port, checked the TX and RX order ;
 - The drone is ON(with full battery), the remote controller too, in the good mode for onboard control, with an iPhone plugged in the remote USB port with the DJI GO app on it and connected to my account.
 
The connection to the serial port works well both on linux and windows, but when try to `request activation` I've got the `--- NO RESPONSE: 374 ---` message coming from DJI_Pro_App.cpp

```
static void * Activate_API_Thread_Func(void * arg)
{
    int retry = 12;
    activate_data_t *temp_data_t = (activate_data_t *)arg;
    from_user_account_data = *temp_data_t;
    while(1)
    {
        DJI_Pro_App_Send_Data( 2, 0, MY_ACTIVATION_SET, API_USER_ACTIVATION,
                   (unsigned char*)&from_user_account_data,
                   sizeof(from_user_account_data) - sizeof(char *),
                   DJI_Pro_Activate_API_CallBack,1000,1);

        usleep(50000);
        sleep(1);
        if(to_user_activation_result == SDK_ERR_NO_RESPONSE)
        {
            printf("--- NO RESPONSE: %d ---\n",__LINE__);

				pthread_mutex_lock(&std_msg_lock);
				std_broadcast_data.activation= 2;
				pthread_mutex_unlock(&std_msg_lock);

            if(p_activate_api_interface)
            {
                p_activate_api_interface(SDK_ERR_NO_RESPONSE);
            }
            activate_api_lock = -1;
            break;
        }
        else if(to_user_activation_result == SDK_ACTIVATE_NEW_DEVICE)
        {
```

If I try some other functionalities, they just don't work too. I've took a look on what was going through the serial port (with all the existing baud-rates) and I've never seen any human readable messages, but the serial port is actually receiving something!

Do you have any ideas of what I'm doing wrong?
thanks in advance

lanyusea, 2016-02-29:
since it doesn't work on both Linux and win

I'm wondering if your USB-UART model works well?

could you find the device on the COM device list of QT GUI?

nesnes, 2016-02-29:
Yes I found it on windows as COM1 an on linux as /dev/ttyUSB0 , and I tested the USB-UART by connecting it too itself TX on RX and I'm receiving the messages that I'm sending. So I think that it is not the problem, it's a classic one coming from adafruit bought one week ago.

lanyusea, 2016-02-29:
so the COM port is shown in QT GUI? not only in Windows setting?

groundmelon, 2016-02-29:
Does your iPhone have internet access?

Are the baudrates are both 230400?

nesnes, 2016-02-29:
Triple yes!
QT GUI can see COM1
Iphone have 4G internet
Baudrate in DJI-Assistant is 230400(set with drone connected in usb)
Baudrate in QT GUI is 230400
Baudrate in config.xml(linux) is 230400

Note that RX: at the bottom left of QT GUI is incrementing!

lanyusea, 2016-02-29:
remove all broadcast data, turn into 115200 and try again.

I'm guessing you choose a half-duplex module

nesnes, 2016-03-01:
ok, I think you have in mind what's happening, I did what you recommended and now the activation is successful and  I can take the control of the drone from the serial port. What settings would you recommend me for having some data broadcast? Should I buy another serial converter?(do you have on to recommend me?) 
Thank you for your help!

lanyusea, 2016-03-01:
If I'm not wrong, the problem is caused by the limitation of data bandwidth.(I apologize I made a mistake, the baudrate should be 230400 instead of 115200, I was too sleepy last night....

There are too many packages your model needs to upload and download, so the ACK of activation result run into package lost.

I suggest you buy a full-duplex USB-UART model, or double half-duplex USB-UART model, one for upstream while the other downstream.

Also, remove unnecessary broadcast data or turn down its frequency is also helpful, like it is not necessary to check the battery status in 100Hz.

nesnes, 2016-03-01:
I'll take a look, thank you!

closed 2016-03-01 by nesnes

lanyusea, 2016-03-01:
u r welcome ;)