dji-sdk/Onboard-SDK #44: Give attitude information to the drone

nesnes, 2016-03-09:
Hi, I have an onboard computer equipped with a camera that is performing a Position/Orientation estimation. My goal is to have the most accurate information of the attitude of the drone, so I'm fusing the camera estimation with the IMU and GPS informations given by the drone. This gives me a quite good estimation, but this information remains in the onboard computer for now...

Is there a way to provide this estimation to the matrice100 to help him determining his horizontal speed? Because this is the only thing I can control to move the drone, I need this speed to be as accurate as possible! Note that I'm talking about a matrice100 without Guidance. The communication with the drone is over serial.

lanyusea, 2016-03-09:
well, as far as I know, the velocity of M100 is calculated by GPS data or IMU fused GPS data.
Therefore it may not reach your expect. 

Maybe you can add an extra sensor for velocity measurement?

nesnes, 2016-03-09:
This extra sensor is the camera that I'm using. So I can fuse the camera, the IMU and the GPS informations to have a more accurate horizontal speed. But this computed data is in the memory of my embedded computer, not in the flight controller of the matrice100. So when I try to command it with velocity commands, the drone compute his attitude using his estimations, not the ones that I want to give him. So I'm stuck with the performance of the IMU+GPS even if I have the datas of IMU+GPS+Camera.
So, is there a way to send my speed estimations to the flight controller ?
If not, I don't see any other solution that controlling the drone in attitude(roll/pitch/yaw), but that's very far from being easy to implement! And the results should be poorer that the algorithm inside the flight controller!

lanyusea, 2016-03-09:
oic. sorry I misunderstood your question at first.

There is no way to do this kind of thing.

But do you think it may be a idea that implement a velocity control logic
by yourself? Since you can control the pitch/roll angle of the drone, and
also you can read the current velocity of it.

nesnes, 2016-03-09:
> If not, I don't see any other solution that controlling the drone in attitude(roll/pitch/yaw), but that's very far from being easy to implement! And the results should be poorer that the algorithm inside the flight controller!

Yes that's was I was thinking, but that's known to be a hard/long task to obtain good performances... Overall this is already implemented in your flight controller. I assume that there is no way for me to send those informations over the CAN bus like Guidance is doing?

However, two linked questions: 
1) I can control the drone in roll/pitch/yaw/thrust , I can get the current yaw/pitch/roll with the quaternion, but how can I get the current thrust value? Not the height/altitude/z-velocity, the thrust value that seems to be ~40 when the drone is hovering.

2) In the simulator, the drone is well simulating the GPS and know perfectly his velocity/position because it think that he is embedding the Guidance system. Is there a way to simulate the drone without the Guidance to have more realistic datas? So to have less performances in the drone's estimations datas? So I'll get a behaviour nearer to the reality!

lanyusea, 2016-03-09:
no you cannot.

1. there is no way to get the throttle value.
and I'm wondering why throttle value is necessary since you have already fused and gotten a reliable velocity data. The input variable is roll/pitch/throttle, the output is vx,vy,vz, the feedback is vx,vy,vz. I think you can build a nested PID loop where x,y inside while z outside. Try to adjust the thrust directly to achieve the goal or an effective way that is find a relation between thrust and acceleration. The thrust value is not necessary as I think, neither the angle of pitch/roll since you have already measured the vx,vy,vz which provides enough degree of freedom.

2. no, there is no guidance inside. The reason why it perfect is just because it is simulated, everything in its environment is perfect. No wind, no noise, no gravity point diverse, it is just like brain in the Matrix.

nesnes, 2016-03-09:
Ok, thanks for those informations and for your time, have a good day! :)

closed 2016-03-09 by nesnes