dji-sdk/Onboard-SDK #46: Onboard SDK serial communication --- NO RESPONSE

sytelus, 2016-04-26:
I can run DJI Assitant and Simulator fine. However when using PureQT sample app or console app, there is no response when I click Activate button. I've confirmed COM port is correctly connected and opened with exact same baud rate as in DJI Assistant. Are there any other requirements? Please help as we can't anything else until this problem is solved.

sytelus, 2016-04-27:
We finally figured out the issue. So apparently I hadn't connected UART to USB board+cable. There was an instruction that this is sold separately but many days ago when I assembled the kit I thought the mainboard was already connected through back USB port so may be instructions was outdated. It was also unbelievable that DJI would left out this critical piece of cable that costs $20 with its $2000 drone - especially when the drone is being marketed as "developer" drone. It's mind boggling but that's is. Apparently, anyone who buys Matrice must search for this UART to USB board on Amazon, purchase it separately and THEN only you can actually talk to your Matrice! It's beyond me why designers decided not to use back USB port that already exists so nicely and connected to the board.

This is what you need to buy from Amazon:
http://www.amazon.com/gp/product/B00EIALP16

lanyusea, 2016-04-27:
because we don't produce the usb-tty module.

closed 2016-04-27 by lanyusea

bossmcfoss, 2017-01-24:
Hi, @sytelus 

Did you mean you had to use the UART to USB cable to connect the Matrice to the PC? I'm troubleshooting some connection issues as well.

Thanks!