dji-sdk/Onboard-SDK #49: Channels assignment for setBroadvastFreq

sytelus, 2016-05-11:
It looks like below function allows to set broadcast frequency for data. However it takes array of size 16, each of which elements probably mean certain broadcast data type. This doesn't seem to be documented anywhere. What is the default channel assignment for Matrice 100 for below function? I want to set broadcast frequency for specific messages (for example, local_position) and set everything else to 0 Hz to remove noise. I can certainly do this from PC Assistant but it would be great to be able to do from APIs at run time. Also when can below function may be called (after activation or after RequestControl)?

`void CoreAPI::setBroadcastFreq(uint8_t *dataLenIs16, CallBack callback, UserData userData)`

It would be also nice to update the API doc with this info.

lanyusea, 2016-05-11:
https://github.com/dji-sdk/Onboard-SDK/blob/3.1/doc/en/OPENProtocol.md

please refer to command 0x00 0x10

sytelus, 2016-05-11:
Thank you! Just wondering if there are any restrictions on when this command can be executed...

lanyusea, 2016-05-11:
@sytelus should be no restriction. As I remember, it is the same as the one in DJI Assistant, by which you can modify the frequency whenever you want.

closed 2016-05-23 by lanyusea

peza8, 2020-04-01:
Hi @lanyusea - This topic was closed on 23 May 2016, and the reference to the OPEN protocol doesn't seem to match the 'Available Broadcast Data' described in the DJI documentation [here](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-telemetry.html). 

I have an M210 and am interested in accessing Relative Position. Where can I set the broadcast frequency of this topic? Thanks.