dji-sdk/Onboard-SDK #51: 一个缓冲区溢出的bug

intijk, 2016-05-22:
**lib/inc/DJI_Type.h**
```
#define API_LOG(driver, title, fmt, ...)                                                       \
    if ((title))                                                                               \
    {                                                                                          \
        int len = (sprintf(DJI::onboardSDK::buffer, "%s %s,line %d: " fmt,                     \
                           (title) ? (title) : "NONE", __func__, __LINE__, ##__VA_ARGS__));    \
        if ((len != -1) && (len < 1024))                                                       \
            (driver)->displayLog();                                                            \
        else                                                                                   \
            (driver)->displayLog("ERROR: log printer inner fault\n");                          \
    }
```

这样做貌似sprintf在目标字符串超过1024的时候就溢出了，程序就会报seg fault, 这个len永远不会返回，这里应该用snprintf.

我这里没有你们的ci，我怕提交了pr以后万一有bug，希望你们check下这个溢出的问题 :)

intijk, 2016-05-22:
这部分代码是用来检测log字符串长度的，超过就报错的意思，而且sprintf会在字符串的结尾加 '\0', 可以把buffer的大小加1，然后用snprintf来限制字串长度防止overflow，然后再检查长度，如果log输出不完整再报错。

intijk, 2016-05-22:
还有类似displayLog 部分的代码是不是应该考虑用fprintf打到stderr去？

lanyusea, 2016-05-23:
snpritf  不是标准库，如果改了将无法跨平台至 stm32

intijk, 2016-05-23:
snprintf  在 c++11是标准了。

不管用什么吧，现在的实现都是一个缓冲区溢出的bug，不考虑改了吗？

lanyusea, 2016-05-23:
buffer size定义在 const size_t bufsize = 1024; 
如果会溢出的话你可以增大这个size。考虑到打印的需求应该不多于1024（一个pkg也没这么长），就设定了1024作为buffer长度。

intijk, 2016-05-23:
1. 用户不一定知道这个啊。 写了一个很长的debug string传进去，然后seg fault了，岂不是一脸懵逼？能提示下也好啊。
2. 既然你都这么说了，要自己设，那就是“撒手不管崩了算你的”策略了，还检查返回值len干啥？

lanyusea, 2016-05-23:
然而目前来看就只有你一个用户有提问，崩了并不会不管，来问总会跟他说的。

intijk, 2016-05-23:
额，这样对待bug不太好吧。。。

那你close吧

closed 2016-05-23 by lanyusea

jackwish, 2016-05-23:
...

summershrimp, 2016-07-18:
...