dji-sdk/Onboard-SDK #54: activate  SDK version did not match

hewen1990, 2016-05-27:
我的M100刚升级到 MATRICE100_FW_V01.03.01.00 
我用了命令行的linux sample, 出现了：activate  SDK version did not match问题。

lanyusea, 2016-05-27:
请升级飞机固件至最新

KellyHwong, 2016-05-31:
我也遇到了这种问题。
我的M100已经升级到MATRICE100_FW_V01.03.01.00了（这个是最新的吧？），使用的Onboard SDK也是最新版本。
在Manifold上编译了其中sample里面的Linux文件夹中的工具，使用--SS load key.txt命令激活，提示以下信息：
activate SDK version did not match
STATUS getDroneVersionCallback,line 252: version ack = 65281STATUS getDroneVersionCallback,line 253: version crc = 0xA6453AACSTATUS getDroneVersionCallback,line 255: version ID = 041D492163STATUS getDroneVersionCallback,line 257: version name = SDK-v1.0 BETA M100-03.01.01.00
请问这是为什么呢？

lanyusea, 2016-05-31:
用ROS试一下吧，cmdline的例程一直无人维护

cloudlj, 2016-06-15:
各位高手，问题解决了么？我也遇到这个问题了，固件版本都换了又换，如何才能连接成功呢???

lanyusea, 2016-07-03:
closed because we have updated all codes/docs about the onboard SDK to solve any possible problem.

please reopen the issue if you still suffer the same problem

closed 2016-07-03 by lanyusea