dji-sdk/Onboard-SDK #59: Quaternion to Eulerian conversion

sytelus, 2016-06-14:
The [programming guide](https://developer.dji.com/onboard-sdk/documentation/ProgrammingGuide/) says conversion code is this:

```
float roll  = atan2(2.0 * (q.q3 * q.q2 + q.q0 * q.q1) , 1.0 - 2.0 * (q.q1 * q.q1 + q.q2 * q.q2));
float pitch = asin(2.0 * (q.q2 * q.q0 - q.q3 * q.q1));
float yaw   = atan2(2.0 * (q.q3 * q.q0 + q.q1 * q.q2) , - 1.0 + 2.0 * (q.q0 * q.q0 + q.q1 * q.q1));
```

However the implementation in DJI SDK does not match above:

```
//Source: http://docs.ros.org/latest-lts/api/dji_sdk_lib/html/DJI__Flight_8cpp_source.html#l00152
00152 EulerianAngle Flight::toEulerianAngle(QuaternionData data)
00153 {
00154     EulerianAngle ans;
00155 
00156     double q2sqr = data.q2 * data.q2;
00157     double t0 = -2.0 * (q2sqr + data.q3 * data.q3) + 1.0;
00158     double t1 = +2.0 * (data.q1 * data.q2 + data.q0 * data.q3);
00159     double t2 = -2.0 * (data.q1 * data.q3 - data.q0 * data.q2);
00160     double t3 = +2.0 * (data.q2 * data.q3 + data.q0 * data.q1);
00161     double t4 = -2.0 * (data.q1 * data.q1 + q2sqr) + 1.0;
00162 
00163     t2 = t2 > 1.0 ? 1.0 : t2;
00164     t2 = t2 < -1.0 ? -1.0 : t2;
00165 
00166     ans.pitch = asin(t2);
00167     ans.roll = atan2(t3, t4);
00168     ans.yaw = atan2(t1, t0);
00169 
00170     return ans;
00171 }
```

lanyusea, 2016-06-14:
em..... I cannot see any difference

only the yaw have different positive direction

uwleahcim, 2016-06-22:
Also note, I believe DJI defines quaternion q0, q1, q2, q3 as w, x, y, z. 

ROS defines quaternion q0, q1, q2, q3 as x, y, z, w.

lanyusea, 2016-06-22:
@uwleahcim  em.... I believe DJI define w, x, y, z as q0 q1 q2 q3, 
ROS define x y z w as q1 q2 q3 q0

uwleahcim, 2016-06-22:
Ah yes, basically check the documentation before assuming parameters are q0, q1, q2, q3. In ROS tf::Quaternion(0, 0, 0, 1) is zero rotation since  tf::Quaternion(x, y, z, w).

lanyusea, 2016-07-03:
closed because we have updated all codes/docs about the onboard SDK to solve any possible problem.

please reopen the issue if you still suffer the same problem

closed 2016-07-03 by lanyusea