dji-sdk/Onboard-SDK #62: SDK

stedeba, 2016-07-01:


stedeba, 2016-07-01:
Using:
fw v1.3.1
SDK version: 3.1
Latest DJI Assistant2
I get:
ERROR activateCallback,line 302: activate SDK version did not match
STATUS getDroneVersionCallback,line 252: version ack = 65281
STATUS getDroneVersionCallback,line 253: version crc = 0xA6453AAC
STATUS getDroneVersionCallback,line 255: version ID = 041D292424
STATUS getDroneVersionCallback,line 257: version name = SDK-v1.0 BETA M100-03.01.01.00

Using Linux CommandLine examples.
Any suggestion or fix?

stedeba, 2016-07-01:
I managed to solve the issue, the bug was on commandline/Linux/conboardsdk/main.cpp...the sdk version was not given to the CoreAPI constructor.

lanyusea, 2016-07-02:
we just updated the code, please use the latest one

lanyusea, 2016-07-03:
closed because we have updated all codes/docs about the onboard SDK to solve any possible problem.

please reopen the issue if you still suffer the same problem

closed 2016-07-03 by lanyusea