dji-sdk/Onboard-SDK #65: error: 'dataflag' was not declared in this scope

zjdavis, 2016-07-04:
Hi, there,

I have just downloaded the Onboard-sdk 3.1.7, when I am compiling, I met the problem:

I went to sample/pureQT directory, and  used the commands as 
```
qmake 
make
```

```
/usr/lib/arm-linux-gnueabihf/qt5/bin/uic djionboardsdk.ui -o ui_djionboardsdk.h
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT -DDJILIB -DDEPENDENCE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -I../../../lib -I../../../lib/inc -I../../../lib/onboardSDK/inc -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o main.o main.cpp
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT -DDJILIB -DDEPENDENCE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -I../../../lib -I../../../lib/inc -I../../../lib/onboardSDK/inc -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o djionboardsdk.o djionboardsdk.cpp
djionboardsdk.cpp:1571:6: warning: unused parameter 'checked' [-Wunused-parameter]
 void DJIonboardSDK::on_btn_wp_start_stop_clicked(bool checked)
      ^
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT -DDJILIB -DDEPENDENCE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -I../../../lib -I../../../lib/inc -I../../../lib/onboardSDK/inc -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o QonboardSDK.o QonboardSDK.cpp
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT -DDJILIB -DDEPENDENCE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -I../../../lib -I../../../lib/inc -I../../../lib/onboardSDK/inc -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o DJI_API.o ../../../lib/src/DJI_API.cpp
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT -DDJILIB -DDEPENDENCE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -I../../../lib -I../../../lib/inc -I../../../lib/onboardSDK/inc -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o DJI_App.o ../../../lib/src/DJI_App.cpp
../../../lib/src/DJI_App.cpp: In member function 'void DJI::onboardSDK::CoreAPI::broadcast(DJI::onboardSDK::Header*)':
../../../lib/src/DJI_App.cpp:95:25: error: 'dataflag' was not declared in this scope
   passData(*enableFlag, dataflag, &broadcastData.gimbal, pdata,
                         ^
make: *** [DJI_App.o] Error 1
```

I turned you don't have any declaration on dataflag when I try use the command
```
grep 'dataflag' -r *
```
the results are like this:
```
lib/src/DJI_App.cpp:  passData(*enableFlag, dataflag, &broadcastData.gimbal, pdata,
lib/src/DJI_App.cpp:  passData(*enableFlag, dataflag, &broadcastData.ctrlInfo, pdata,
```

as you declared the passData, 
```
inline void passData(uint16_t flag, uint16_t &enable, void *data, unsigned char *buf,
    size_t datalen, size_t &offset)
```

you should add a declaration for dataflag in the file.

rohitsantdji, 2016-07-04:
Has already been fixed in latest bugfix commit. Clone from 3.1 branch, not the 3.1.7 tag to get the fix.

closed 2016-07-04 by rohitsantdji

rohitsantdji, 2016-07-05:
Updated the 3.1.7 tag to include the bugfix.