dji-sdk/Onboard-SDK #66: DJI Simulator API

sytelus, 2016-08-19:
Is there anyway to access information in DJI simulator? The DJI simulator is very limited (no environment, collisons, sensors etc) so we want to try out Gazebo or something like that. All we currently need is ability to get position and orientation that is in DJI Simulator. This we can then feed in to Gazeebo so that we use hardware-in-loop part. So if any of below things are possible, we can possibly make it work:

1. Ability to query current position and orientation in DJI simulator.
2. Ability to trigger hardware-in-loop simulator mode using some API.

I won't mind trying out out something experimental/beta so we can keep going :)

a2hill, 2016-08-19:
I believe simulator mode can be triggered from the mobile sdk, so you could build an app that triggers it that way and then receive position updates over serial via the onboard api

rohitsantdji, 2016-08-23:
@a2hill is correct. The Mobile SDK allows triggering the simulation programmatically through an API. The Onboard SDK allows you to query drone state. The [docs](https://developer.dji.com/onboard-sdk/documentation/appendix/index.html#flight-data) should help with the flight data part.

closed 2016-08-23 by rohitsantdji