dji-sdk/Onboard-SDK #67: Lock m_msgLock before returning the broadcastData ?

flm8620, 2016-09-22:
[https://github.com/dji-sdk/Onboard-SDK/blob/edf5f76b2b8d37878126ec1f4d7b3a3a0dfe092d/lib/src/DJI_App.cpp#L45](url)
I'm not so sure if I have missed something, I think there is a problem of a shared variable between main thread and read thread.

In DJI::onboardSDK::CoreAPI::getBroadcastData()
The main thread get `broadcastData` without mutex protection. 
This value may be changed by read thread in 
`void DJI::onboardSDK::CoreAPI::broadcast(Header *protocolHeader)`
in which the read thread calls `serialDevice->lockMSG()` before overwriting `broadcastData`.

flm8620, 2016-09-26:
Actually all members in CoreAPI (such as broadcastData, broadcastFrameStatus) and LinuxThread::stopCond aren't protected by mutex.

lanyusea, 2016-11-15:
well, in our original design at first, the core library is cross-platform, which means we have to support some embedded platform like STM32 and Arduino, where there is no multi-threads.

so at beginning the broadcast receving->updateing->reading is in serial and single thread.

but it seems things have been changed a lot in our recent releases.

Thanks for your report, we will work on this mutex issue later.

flm8620, 2016-11-15:
Got it, Thanks.

rohitsantdji, 2016-11-15:
Some additional clarification to add to @lanyusea's answer -  a fix for this issue has been planned for the next release, we're looking at a December timeline.

closed 2017-01-12 by lanyusea