dji-sdk/Onboard-SDK #68: Data frames are not received in order

alexandrubichis, 2016-09-26:
Hello,
	
I'm working with Mobile Onboard SDK communication using Data Transparent Transmission, to send a data frames from Mobile Device to an Onboard Embedded System through A3 with Lb2. My issue is that my data frames (6 bytes the size of data frame) are not received in order (I've created a sequence number in my data frame that is incremented on mobile device when sending the message). I've checked the protocol frame, and the frame sequence number from header it is received correctly. Can you check please if the the issue is on Mobile SDK or on Onboard SDK.
	
Thank you

lanyusea, 2016-11-15:
Hi alex,

sorry I just noticed this post.

We have some similar applications from our developers, and there is no similar issue reported (at least, to me).

1. what is the version of your A3 firmware?
2. can it be duplicate every time?
3. What is the version of your Mobile SDK? Are you using Android or iOS?

and to be more clear,

did you mean,
1. the sequence number of protocol header is correct
however,
2. the sequence number, which is assigned by yourself in data bytes is disordered?

alexandrubichis, 2016-11-16:
Hello,

A3 firmware version is 1.3.0 and mobile sdk 3.3. I've tested with both platforms and I have the same problem. I'm sending at every 20 ms a new message(6 bytes lenght)  from my mobile with my internal sequence incremented. Even if I increase from 20 ms to 2 seconds I have the same issue.

Another issue that I found is that on last A3 firmware version (1.5.0) communication from Onboard Embedded System to Mobile is not working. I've tested with both 3.1.9, 3.1.8 onboard-sdk version and 3.4,3.3 mobile sdk (iOS,Android). Transmission from mobile to OES is working fine.  

Thank you

lanyusea, 2016-11-17:
Hi Alex,

What is your onboard device? Is it a embedded system like stm32?
We have confirmed that there is a bug in sending data from onboard to
mobile. Someone just send the pointer to the mobile device instead of the
content. We just kicked his ball last night and it will be fixed in the
next release.

alexandrubichis, 2016-11-17:
Hi,

Yes I'm using a stm32.

alexandrubichis, 2017-01-12:
Hello,

I have a new issue with latest A3 fmw (1.6.0). Data transparent
transmission from mobile is received incorrectly.

We implement an application on an OES device using OnboardSDK.
It receives some data from an android device running MobileSdk through Data
Transparent transmission.
With A3 fw version 1.3.0.00 the OES receives the data correctly.

When changing to A3 fw version 1.6.0.10 with the same code the data is
received to OES, but it can't be recognized.
The header of the DTT packet is the same, but the payload is nothing usable.

Here is an example of 3 bytes data packet received from MobileSDK:

A3 1.3.0.00:
aa 15 00 00 00 00 00 00 20 00 11 de 02 02 01 01 00 81 30 1f 5b

A3 fw 1.6.0.10:
aa 16 00 00 00 00 00 00 b6 81 aa ee 02 02 20 73 44 20 af aa 9a b7

Could it be that the data is transmitted encrypted on the new fw 1.6.0.10?

Note:
1. While changing code to work on fw 1.6 we updated to the newest
OnboardSDK 3.2 and MobileSDK for Android 3.5.
 The A3 with fw version 1.3 receives the data correctly.

2.  When using A3 1.3.0.00 the activation of OnboardSDK is requested with
  "versionA3_31 = (MAKE_VERSION(3, 1, 100, 0));" (successfully),
 while when using A3 1.6.0.10 the activation of OnboardSDK is requested
with:
 "versionA3_32 = (MAKE_VERSION(3, 2, 100, 0));" (also successfully).

lanyusea, 2017-03-10:
hi @alexandrubichis sorry to reply such late.
is the bug still exist in our latest firmware / sdk code ?

if so, please open a new issue ticket to report.
It is really hard to track your reply among all issues.

closed 2017-03-10 by lanyusea