dji-sdk/Onboard-SDK #69: Battery level always zero on A3

maciekd, 2016-11-01:
Hi,

According to DJI support the battery level value is not supported on the A3 and will always return zero. I've verified this on my A3 with S900 frame and non-intelligent battery.

Would it be possible to include this in a future version? In fast, what I'd really like would be the actual voltage value from the battery (as shown in the DJI Assistant).

chukcha2, 2016-11-09:
Are you referring to 'battery info'? According to documentation of CMD Set 0x02 (Push Data CMD Set), CMD ID 0x00 (Flight Data), "item presence byte" for A3 contains "flag of battery info" in bit 12. Are you referring to this feature or something else?

maciekd, 2016-11-09:
Yep. I assume this value corresponds to 

`BatteryData battery;`

(a byte) in the `BroadcastData` struct.

That value is always zero on the A3 according to my testing (and DJI support).

lanyusea, 2016-11-15:
to clarify,

the battery info is only supported for DJI intelligent battery (well, I don't like this name actually), which has a serial port to talk with the drone.

for custom batteries, the A3 can only measure its voltage, which means it is impossible to calculate how many percentages left. Currently, we only use the voltage to trigger low-battery actions, whose functionality will be somehow overlapped if we publish the voltage left by SDK and make the developers decide what to do when the battery is quite low, because 1. some problem by our system architecture design if the low-voltage action can be decided by two ways 2. the low-battery action is coercive, during which the SDK logic has to be exit 3. not easy to specify whose duty when the drone crash because of low voltage.

maybe we can do some change in the future, but I cannot guarantee it.

maciekd, 2016-11-16:
I understand completely, thanks, closing this one.

closed 2016-11-16 by maciekd