dji-sdk/Onboard-SDK #70: Reading ALL Remote Control channels

nikonikolov, 2016-11-09:
Looking though the code it seems like the struct read from the DJI autopilot returns `RCData` struct which only contains the roll, pitch, yaw, throttle, mode and gear signals. There is another struct `VirtualRCData` which contains all channels but it does not seem to be used anywhere. How can I read all the channels from the remote control. Apparently this is possible according to this https://www.youtube.com/watch?v=_9VR9IQP7QM

rohitsantdji, 2016-11-09:
Hi Nikolay,

Those six values are the only RC Data fields available through the [open protocol](https://developer.dji.com/onboard-sdk/documentation/appendix/index.html#flight-data). The `VirtualRCData` struct is used for setting data, not for getting it, when in [Virtual RC mode](https://developer.dji.com/onboard-sdk/documentation/introduction/virtual-rc-protocol.html). 

The video most likely uses information from other parts of the SDK (e.g. flight status for LAND, OFF display and battery percentage from the BroadcastData).

closed 2016-11-09 by rohitsantdji

jasperza, 2016-11-11:
I made that video. I dont use the SDK for the onboard, I wrote my own code
for Arduino.  The DJI SDK is not compatible with arduino because it uses
C++ functionality not available on arduino,  like multi-threading.

I  wrote my code from nothing, it consists of a class which interprets the
binary data directly from the DJI serial UART port onboard.I adapts to
different messages being sent. It processes the data byte by byte, and bit
by bit.

rohitsantdji, 2016-11-11:
@jasperza That's very cool! We do provide a single-threaded DJI Onboard SDK implementation for STM32, but for the computing budget of an Arduino the smart answer in many cases is to implement your own protocol parser that only deals with the messages you're interested in.

Thanks for sharing the video!

nikonikolov, 2016-11-12:
@jasperza Thanks a lot for sharing!

jasperza, 2016-11-13:
Pleasure!
Also,  every time the class gets a message it checks the data against the
data in the previous message and if something has changed, performs an
action, eg releases a servo.

The class also prepares and outputs data back to the drone for flight
control,  so we programmatically get the matrice to take off and hover at a
certain height.

We have control over all the directions of travel,  we communicate with our
server over our own protocol to set the target height ,  but once set,  all
of the control is fully on board on the arduino.

The arduino monitors the rc signals from the drone over  the serial port
and if the user wants to control the drone using the sticks it releases
control to the user.  After one second of no user input, the arduino takes
over again.

This means that the operator can make quick, small adjustments to one or
two axes without changing the mode switch.

camilojimo, 2017-12-04:
Hi, I'm starting to develop my own app using osdk stm32 discovery and A3 flight controller. It has come to my attention that to run the code from the samples (https://github.com/dji-sdk/Onboard-SDK/tree/3.1/sample/STM32) using  MDK-ARM (Keil uVision) toolchain it is requairing a license because of the size of the code. I'm wondering if @nikonikolov were successful i wold like to know how did you set up your model. also, @jasperza, I'm open in terms of software to cominicate with the flight controler and using Arduino is a possibility for me, if its possible could you give me more details on how you did your set up. I want to take control of yaw with an external parameter and maintain control of everything else with the controller. I've used arduino to read what is coming out from the flight controller but at this point is only data that doesn't give me any information because i don't know when to start reading and when to stop.
if you @jasperza  could explain me in more detail how to ('interprets the binary data directly from the DJI serial UART port onboard.I adapts to different messages being sent. It processes the data byte by byte, and bit by bit.') i would very much appreciate that 

Thanks !

jasperza, 2017-12-05:
You have to read each byte that comes from the serial port and make a
decision what to do with it and how that changes the remainder of the
message. There's no easy way to do it. It's a few thousand lines of arduino
code. Just get the message from the drone, identify each type of sub
message,  and process it depending on message type.

nikonikolov, 2017-12-05:
@camilojimo I used a Raspberry Pi and developed on Linux. Never actually ran the samples code and did not need any license for my custom app. I would recommend switching to Linux toolchains as this will save you a lot of headaches.