dji-sdk/Onboard-SDK #71: Flight Controller Serial Number

a2hill, 2016-11-12:
Not sure if I'm overlooking it, but is there no way to query the serial number from the Flight Controller?

lanyusea, 2016-11-13:
1. You want to do it by using Onboard SDK?
2. if 1 is true, do you want a UID to separate different drones or you really want to check the serial number?

a2hill, 2016-11-13:
1. Yes I want to use the onboard SDK to do it.
2. UID would be fine

lanyusea, 2016-11-13:
As I remember, there is some bytes describing drone's UID in the ack of (0x00, 0x00): Get Protocal Version. You can printf the ack result to check.
We use it to allow the code separate different drones when swarm.

a2hill, 2016-11-13:
Ahh got it, thanks

lanyusea, 2016-11-16:
@a2hill  does it satisfy your requirement?

a2hill, 2016-11-16:
@lanyusea I've been looking at VersionData returned from getDroneVersion and unfortunately it seems to be empty. Do I need to send a 0x00 0x00 and parse the ACK manually?

rohitsantdji, 2016-11-16:
@a2hill what flight controller and firmware are you using?

a2hill, 2016-11-16:
M100 with firmware 1.1.0.10+

rohitsantdji, 2016-11-16:
We will be releasing a hotfix soon that should take care of this issue. For now, please use the void-return-type getDroneVersion with your own callback as an argument (you can look at getDroneVersionCallback in DJI_API.cpp and implement something very similar, except instead of printing out values using API_LOG you can execute your own logic).

a2hill, 2016-11-16:
Great thanks this worked as far as getting the data! So can the version ID returned be used as a UID of the aircraft?

rohitsantdji, 2016-11-16:
Great! yes, it can.

closed 2016-11-16 by a2hill