dji-sdk/Onboard-SDK #72: GPS health always zero

maciekd, 2016-11-15:
Hi,

On my A3 I always get zero for positional health value (BoadcastData -> pos -> health). I've made dozens of tests. Always zero. In the simulator I always get value 4.

Is this a known problem? Has anyone else had this problem?

Thanks in advance.

lanyusea, 2016-11-15:
If you can get 4 in the simulator, there should be no problem.

1. What does the DJI GO say? Does it show Safe Fly(GPS) in the top bar when there is not value?
2. Did you open the pos publication in the DJI Assistant?
3. What is your firmware version?
4. Can you get the GPS position data successfully and correctly?

maciekd, 2016-11-16:
2. Not sure what you mean by this. Are you talking about the Baud & Data Transmission Rate for Position?
3. 1.3.0.00 on A3

Will get back to you about 4 when I've been flying outdoors. I have no Lightbridge controller right now so can't check DJI GO.

lanyusea, 2016-11-16:
@maciekd 
I mean if the top bar of DJI go is like this or not:
http://cdn5.myfirstdrone.com/wp-content/uploads/2016/03/IMG_0190.jpg

and if you're testing indoor.
then 0 value is quite normal =。=

because the GPS signal is quite weak in the door, so it is not reliable.

maciekd, 2016-11-16:
Very sorry, mixed the order of the questions. I meant I wasn't sure what you mean by question no. 2 ("Did you open the pos publication in the DJI Assistant").

lanyusea, 2016-11-16:
I mean click the box before GPS information in the SDK page.

But I believed you have already done this from your previous replies.

maciekd, 2016-11-16:
Yep. 

But I have to say, I've flown a lot outdoors (maybe a totalt of 20 full days) and I've never seen the health value to change to anything else than 0.

When I fly outdoors next time I'll verify that the GPS position I get is good and will get back to you.

lanyusea, 2016-11-17:
@rohitsantdji 
hi rohit, can you guys duplicate this issue? 
> flying outdoors with GPS but GPS health is 0

maciekd, 2016-11-17:
FYI, here's how I set up my BroadcastData:

`byte freq[] = new byte[16];`
`freq[0] = BROADCAST_FREQ_1HZ;   // Timestamp`
`freq[1] = BROADCAST_FREQ_50HZ;  // Attitude Quaternions`
`freq[2] = BROADCAST_FREQ_50HZ;  // Acceleration`
`        freq[3] = BROADCAST_FREQ_50HZ;  // Velocity (Ground Frame)`
`        freq[4] = BROADCAST_FREQ_50HZ;  // Angular Velocity (Body Frame)`
`        freq[5] = BROADCAST_FREQ_10HZ;  // Position`
`        freq[6] = BROADCAST_FREQ_10HZ;  // GPS`
`        freq[7] = BROADCAST_FREQ_1HZ;   // RTK`
`        freq[8] = BROADCAST_FREQ_1HZ;   // Magnetometer`
`        freq[9] = BROADCAST_FREQ_50HZ;  // RC Channels Data`
`        freq[10] = BROADCAST_FREQ_1HZ;  // Gimbal Data`
`        freq[11] = BROADCAST_FREQ_1HZ;  // Flight Status`
`        freq[12] = BROADCAST_FREQ_1HZ;  // Battery Level`
`        freq[13] = BROADCAST_FREQ_1HZ;  // Control Information`

Previously I ran with RTK and Gimbal Data at 0 Hz.

rohitsantdji, 2016-11-17:
We have not seen this issue in our tests with the M600, but we can try and reproduce this with a standalone A3 on a custom frame (@maciekd this is your setup right?)

maciekd, 2016-11-17:
That's exactly right! Im running an A3 on a S900 frame. 

Thanks for helping me out with this.

maciekd, 2016-11-25:
Hi guys,

Just checking back in, any news on this?

Best Regards

rohitsantdji, 2016-12-06:
We did a few tests and were unable to reproduce this issue. We get GPS health changing from 0 -> 3 -> 4 as per changes in satellite configurations.

maciekd, 2016-12-07:
Thanks for looking into this.

Since your reply I did some more research and found the issue:

When powering on our system the A3 controller boots first. At this point it normally has good GPS reception and blinks green. As soon as our onboard system boots a high speed USB 3.0 hub is powered on. I've now confirmed that this USB-hub, when under heavy load (as it always is in our system), completely kills the GPS signal. 

In my previous testing I saw that the A3 was blinking green and then checked our system and got gps health 0. Between these two events the USB-hub had started and always managed to block the GPS signal completely - and that's why I though the A3 always reported GPS health zero.

Sorry for bothering you guys with this.

closed 2016-12-07 by maciekd

meyerj, 2017-11-17:
Even though it was not an SDK issue, we have exactly the same problem with USB 3.0 interference with a [Matrice 100](https://www.dji.com/matrice100) with the standard GPS and a [DJI Manifold](https://www.dji.com/manifold). As soon as a USB 3.0 device is connected, the GPS signal is blocked completely, even with no electrical connection at all between the Manifold and other systems (separate power supply, tried with different USB devices). Could you solve the problem somehow, or is this a known issue?

maciekd, 2017-11-17:
It's a tricky problem but possible to solve. A lot of EMI screening and distance to the GPS is key.

Our landing page has a video rolling (if you are on desktop) that has one temporary solution we used:
https://www.everdrone.com. Notice the ridiculous distance to the GPS.

(video only visible when you're on desktop)