dji-sdk/Onboard-SDK #73: Broken link in documentation uAvionix ADS-B receiver with DJI Onboard SDK page

BlueTower, 2016-11-16:
On this page: https://developer.dji.com/onboard-sdk/documentation/sensor-integration-guides/ping/README.html

The link in the first paragraph to "pingRX" returns a 404 page. You can instead link to this page, which works: https://www.unmannedsystemssource.com/shop/atc-devices/pingrx-ads-b-receiver/

rohitsantdji, 2016-11-17:
Thanks for pointing this out! Unfortunately the unmannedsystemssource link seems to redirect to a wordpress setup page - we'll find a stable link and replace it in the docs.

BlueTower, 2016-11-21:
I think it might have been a single incident. They seem to be back up now. You could also link to the Amazon website: https://www.amazon.com/uAvionix-UAX-90005-01-pingBuddy-Wi-Fi-Receiver/dp/B01IYHIYQM/ref=sr_1_1?ie=UTF8&qid=1479746774&sr=8-1&keywords=pingbuddy

rohitsantdji, 2016-12-06:
Fixed last week. Thanks!

closed 2016-12-06 by rohitsantdji