dji-sdk/Onboard-SDK #74: Can't activate or read data from M600

TheJerms, 2016-11-24:
Followed the instructions on the Onboard SDK website, but once I confirm the port is open and hit activate in the Qt example, nothing happens. Nothing also happens when I hit obtain control. Any advice would be appreciated.

Other things to note:
- OES is a Windows laptop running Qt
- Baud rates are synced to 115200 through a USB port
- [https://postimg.org/image/uajttz06d/](url)
- Would like to read flight telemetry data from the M600

lanyusea, 2016-11-24:
the baudrate in your screenshot is 230400 instead of 115200

TheJerms, 2016-11-24:
* Oops I must have changed it back to 230400 after testing. It gives the same result with 115 200 on both the assistant and the GUI.

lanyusea, 2016-11-24:
em....  could you check if there are data stream sent from A3? by using some software like serial port monitor maybe this one https://sourceforge.net/projects/realterm/

TheJerms, 2016-11-24:
Used the Free Device Monitoring Studio, I see the program writing to the port but not reading anything. Also this is what the wiring looks like on the A3 (not sure if there is an issue here): [wiring](https://postimg.org/image/duay7z58j/)

TheJerms, 2016-11-24:
* Regarding the previous image:
White = RX
Green = TX
Black = GND

lanyusea, 2016-11-24:
if there is no data, it seems to be a hardware config issue.

the pin of API (from left to right) is: VCC GND TX RX 
   (the figure in our document is actually upside down compared with the real situation
  which means you should connect GND to the GND of your USB-TTL, and connect the TX to your RX, connect the RX to your TX

TheJerms, 2016-11-24:
Oh I see. It's working now! Thank you!

lanyusea, 2016-11-24:
:)

closed 2016-11-24 by lanyusea