dji-sdk/Onboard-SDK #75: osdk-wrapper/src/LinuxMobile.cpp:33: undefined reference to `CartesianFrame::CartesianFrame(Eigen::Matrix<double, 3, 1, 0, 3, 1>)'

eastmoutain, 2016-11-25:
when compiling the SDK, the link error show as the title

nikonikolov, 2016-11-27:
I have the same problem. Builds on Ubuntu 16.04 with gcc 5.4.0 but does not build on Raspberry Pi with gcc 4.9.2.

lanyusea, 2016-11-28:
it seems you need to link with Eigen

rohitsantdji, 2016-11-28:
From the Release Notes [page](https://developer.dji.com/onboard-sdk/documentation/appendix/releaseNotes.html#backward-compatibility) in the documentation - _Note that you will be unable to build the Linux samples on platforms other than Ubuntu 16.04/14.04 on x86_64 or Ubuntu 14.04 on ARM (32-bit). This is due to new feature dependencies; this behavior will change in a future release._

Since we provide pre-compiled libraries for the precision trajectory missions, you cannot link to the libraries if you aren't building on the architectures we provide libraries for. In the next couple of weeks, we will release a new version of the SDK that makes this linking optional, so that you may use the Linux Sample on any architecture of your choice, albeit without precision trajectory features. 

If you want to get started in the meantime, you can choose one of two options - 
1. Checkout the [3.1.8](https://github.com/dji-sdk/Onboard-SDK/releases/tag/3.1.8) release from the Github Repo and use that until the new version is released
2. Use the [ROS](https://github.com/dji-sdk/Onboard-SDK-ROS) sample from the 3.1.9 release.

closed 2016-11-28 by rohitsantdji