dji-sdk/Onboard-SDK #76: Segmentation fault

aznshodan, 2016-11-26:
I'm following the instructions here: 
https://developer.dji.com/onboard-sdk/documentation/github-platform-docs/Linux/README.html

I ran the script using this command: 
`./scripts/build —-build-type Debug -—clean true --lidar-logging OFF
`

But I get segmentation fault. This is what I got:

```
[ 72%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/LinuxMobile.cpp.o
[ 75%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/LinuxCamera.cpp.o
[ 77%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/ReadUserConfig.cpp.o
[ 80%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/LinuxCleanup.cpp.o
[ 83%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/LinuxWaypoint.cpp.o
[ 86%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/LinuxFlight.cpp.o
[ 88%] Building CXX object osdk-wrapper/CMakeFiles/djiosdk-wrapper.dir/src/LinuxSetup.cpp.o
Linking CXX static library ../libs/libdjiosdk-wrapper.a
[ 88%] Built target djiosdk-wrapper
Scanning dependencies of target djiosdk-linux-sample-nonblock
[ 91%] Building CXX object sample/Linux/Non-Blocking/CMakeFiles/djiosdk-linux-sample-nonblock.dir/src/main.cpp.o
[ 94%] Building CXX object sample/Linux/Non-Blocking/CMakeFiles/djiosdk-linux-sample-nonblock.dir/src/LinuxInteractiveNonBlocking.cpp.o
Linking CXX executable ../../../bin/djiosdk-linux-sample-nonblock
collect2: error: ld terminated with signal 11 [Segmentation fault]
make[2]: *** [bin/djiosdk-linux-sample-nonblock] Error 1
make[1]: *** [sample/Linux/Non-Blocking/CMakeFiles/djiosdk-linux-sample-nonblock.dir/all] Error 2
make: *** [all] Error 2
```

I'm using Jetson TX1 with ubuntu 14.04.

lanyusea, 2016-11-26:
quite strange to see the segmentation fault when compile.....

http://stackoverflow.com/questions/25371372/collect2-error-ld-terminated-with-signal-11-segmentation-fault-core-dumped

aznshodan, 2016-11-26:
I just realized that this could be my fault. Jetson TX1 has aarch64 but it seems like onboard sdk only support 32bit arm.

closed 2016-11-27 by aznshodan

rohitsantdji, 2016-11-28:
@aznshodan The only issue with the linking here is the precompiled libraries we provide (precision trajectory missions) - they are only built for 32-bit Ubuntu 14.04 on ARM (and 64-bit Ubuntu 16.04 on x86).  In the next release, we will be making it optional for the Linux sample to link with those libraries. You should be able to run all other functionality of the SDK without any issues once that happens.

aznshodan, 2016-11-29:
ah i see. thank you