dji-sdk/Onboard-SDK #77: Failed to obtain control. Your RC mode switch is not in mode F.

aznshodan, 2016-11-27:


lanyusea, 2016-11-27:
the mode bar is on the top of your left hand, is it be put into the F position?

aznshodan, 2016-11-27:
if you meant top left of the controller, then yes. Also I was able to see the mode change on DJI Assistant 2 as I changed the mode on the controller.

lanyusea, 2016-11-27:
is it in F when you power on the drone? if so, you need to move it away and then back.

aznshodan, 2016-11-27:
I did that as well

rohitsantdji, 2016-11-28:
The "mode F" error message is not completely reliable on ARM devices - sometimes, we have seen the program actually error out at an earlier stage (activation for example) but still proceed to the obtain control step where it errors out with this message. This is becasue of the way the serial driver is implemented works differently with USB-Serial adapters and native serial ports. A few posible causes:

1. Did you change the baud rate in both the SDK page on DJI Assistant 2 as well as in UserConfig.txt?
2. Are your connections correct?
3. Have you enabled API control?

aznshodan, 2016-11-29:
1. Yes I changed it in all of them.
2. I believe so
3. I marked the check box in DJI assistant 2

I wasn't sure whether my connection was correct until I ran the Onboard-SDK-ROS version which was able to take control and run the propellers. So I can always resort to using the ROS version but I'd prefer using linux version. 

Is there anything specific I need to do regarding the serial port on my jetson tx1? I tried using the command 
`setserial /dev/ttyTHS2 baud_base 115200
`and 
`stty -F /dev/ttyTHS2 115200
`
which unfortunately didn't do the trick.

rohitsantdji, 2016-11-29:
My hunch is that the program might be failing at activation. We have not tested this on the TX1, so unfortunately we cannot offer any advice on the specifics of the device. Can you try the same thing with a USB-Serial adapter if you have one lying around? 

Some other routine things to check:

1. Is the UserConfig.txt file correctly populated with ID, Key and drone version?
2. Is DJI GO connected and powered on?

Screenshots of the complete terminal log as well as of your setup might be helpful.

rohitsantdji, 2016-12-06:
@aznshodan can you confirm that a pull from the latest repo fixes the issue?

aznshodan, 2016-12-09:
@rohitsantdji  Yes it doesn't get stuck anymore! The only problem is unlike the sdk-ros, the propellers don't rotate when i run the linux sample. This seems to be some mistake on my end.

closed 2016-12-15 by rohitsantdji

newtoniant, 2017-03-14:
@rohitsantdji here is my test results:

With a fresh pull of the OSDK I tried both a USB serial device (FTDI232) which I use on my desktop linux with success and GPIO serial.  

I get two different results depending on the baud rate I set in the Assistant2 and UserConfig.txt.  But I get identical results when using USB Serial (ttyUSB0) and GPIO serial (ttyS0) at the same baud rate.

With 230400 baud I get:
```
STATUS init,line 51: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS init,line 60: ...Serial started successfully.

Attempting activation..
Automatic activation successful.
Failed to obtain control.
Your RC mode switch is not in mode F. (Is the RC connected and paired?)
This program will exit now. 
```

With 115200 baud I get: 
```
Attempting activation..
Your activation did not go through. 
Your SDK version in User_Config.h does not match the one reported by the drone. 
Please correct that, rebuild (do a `make clean` first) and try again.
This program will exit now. 

```

Fresh git clone, both USB Serial and GPIO serial fail to run.  I updated my app id and key prior to running the sample.  The raspberrypi-support branch has versionA3_31 as the version.  I tried that as well as versionM100_31 that I use on my desktop app.  Both gave the same results.  I am using an M100.

crvogt, 2017-04-12:
Hi aznshodan,
do you have a picture or can you confirm the pins you're connected to on J17? I'm suffering similar issues and struggling to confirm that I'm even connected to the tx1 appropriately.

aznshodan, 2017-04-13:
It's definitely J17. The issue was on the driver side. I can't provide a picture since I'm not using jetson on my drone at the moment.

crvogt, 2017-04-13:
On the driver side? Also, in case you recall, based on the pinout, these are the connections I've made. Perhaps you recall if this was the same with yours?

ground - connected
CTS
USB +5v
TXD - connected
RXD - connected
RTS

aznshodan, 2017-04-16:
Yep that looks good. The UART-CAN2 cable that came with DJI has color coded wires; the colors are shown below:

GND - brown (connected) 
CTS - 
USB - 
TXD - orange (connected)
RXD - red (connected)
RTS -

pavankumarbn, 2018-04-25:
I am facing "failed to read from serial device. the on board sdk is not communicating with your drone."
"Vehicle initialization failed". Matrice 100, Platform: Ubuntu 14.04LTS , ROS indigo, Onboard intel NUC computer. any suggestions how to resolve the issue. 

1. I set-up intel nuc onboard computer with ROS - Indigo on Ubuntu 14.04LTS.
2. Using USB to TTL I am connecting from onboard to M100 UART CAN 2.
3. Firing up DJI assistant 2 to setup baud rate.
4. Added App_Id and Enc_Key to dji_sdk launch file.
5. Trying to run sample examples of the SDK.