dji-sdk/Onboard-SDK #79: lsb-release error, running cmake on Raspbian Jessie

spacemig, 2016-12-05:
I just cloned the Onboard-SDK to a Raspberry Pi with the latest Raspbian Jessie.  I got these errors when I run the build script:

```
./scripts/build —-build-type Debug -—clean true --lidar-logging OFF

...

cat: /etc/lsb-release: No such file or directory
CMake Error at osdk-wrapper/CMakeLists.txt:71 (if):

...

cat: /etc/lsb-release: No such file or directory
CMake Error at sample/Linux/Blocking/CMakeLists.txt:70 (if):
```

The file /etc/lsb-release in fact does not exit. But /etc/os-release does exist. 

What would be the right fix?

lanyusea, 2016-12-05:
what about modify the makefile according to your system by replace lsb-release by os-release?
https://github.com/dji-sdk/Onboard-SDK/search?utf8=%E2%9C%93&q=lsb-release&type=Code

rohitsantdji, 2016-12-06:
The lsb-release command is used in Onboard SDK 3.1.9 to decide which version of the precision missions library to link against. Currently (v3.1.9 and v3.1.9.1), running the Linux sample without precision missions is not supported, so you will be unable to build the samples on your Raspbian Jessie setup. Migrating CMake to the os-release command won't help with this core issue. 

See Issue #75 . Reproducing parts of that answer here:

From the Release Notes page in the documentation - _Note that you will be unable to build the Linux samples on platforms other than Ubuntu 16.04/14.04 on x86_64 or Ubuntu 14.04 on ARM (32-bit). This is due to new feature dependencies; this behavior will change in a future release._

Since we provide pre-compiled libraries for the precision trajectory missions, you cannot link to the libraries if you aren't building on the architectures we provide libraries for. In the next couple of weeks, we will release a new version of the SDK that makes this linking optional, so that you may use the Linux Sample on any architecture of your choice, albeit without precision trajectory features.

You can pursue one of two options until the new version is released:

1. Clone the [v3.1.8](https://github.com/dji-sdk/Onboard-SDK/tree/3.1.8) release, navigate to the `sample/Linux` and use `make` to build it, and `./onboardsdk -interactive` to run it
2. Install ROS and use the [ROS](https://github.com/dji-sdk/Onboard-SDK-ROS) version of Onboard SDK 3.1.9.

closed 2016-12-06 by rohitsantdji