dji-sdk/Onboard-SDK #80: What's the <BroadcastData.CtrlInfoData.mode;> indicate?

trickyMan, 2016-12-13:
What's the <BroadcastData.CtrlInfoData.mode >indicate in the Onboard SDK? I can not find out any information from the docs... 

And my real question is:
I'm seeking the <Flight Mode> in the iOS SDK which is sent from Onboard SDK. I need this mode/status to do some stratage . But I can not find out the approach to get this status. 

And I only get the <BroadcastData .CtrlInfoData.mode> in the Onboard SDK. It seems that this mode indicates the flight mode which is described in the iOS SDK...

And this is the definition of the Flight Mode:
[Flight Mode Definition in iOS SDK](https://github.com/dji-sdk/Mobile-SDK-iOS/blob/c0aba6dd300465e0a0993d2a7dcf9c8333c6f742/API%20Reference/Docset_version/DJIMobileSDK-for-iOS.docset/Contents/Resources/Documents/docsets/DJISDK.docset/Contents/Resources/Documents/Enums/DJIFlightControllerFlightMode.html)

Waiting for you response.
Thank you.

lanyusea, 2016-12-13:
It is in the [Appendix](https://developer.dji.com/onboard-sdk/documentation/appendix/index.html)

```
typedef enum 
{
  ATTI_STOP                    = 0,
  HORIZ_ANG_VERT_VEL_YAW_ANG   = 1,
  HORIZ_ANG_VERT_VEL_YAW_RATE  = 2,
  HORIZ_VEL_VERT_VEL_YAW_ANG   = 3,
  HORIZ_VEL_VERT_VEL_YAW_RATE  = 4,
  HORIZ_POS_VERT_VEL_YAW_ANG   = 5,
  HORIZ_POS_VERT_VEL_YAW_RATE  = 6,
  HORIZ_ANG_VERT_POS_YAW_ANG   = 7,
  HORIZ_ANG_VERT_POS_YAW_RATE  = 8,
  HORIZ_VEL_VERT_POS_YAW_ANG   = 9,
  HORIZ_VEL_VERT_POS_YAW_RATE  = 10,
  HORIZ_POS_VERT_POS_YAW_ANG   = 11,
  HORIZ_POS_VERT_POS_YAW_RATE  = 12,
  HORIZ_ANG_VERT_THR_YAW_ANG   = 13,
  HORIZ_ANG_VERT_THR_YAW_RATE  = 14,
  HORIZ_VEL_VERT_THR_YAW_ANG   = 15,
  HORIZ_VEL_VERT_THR_YAW_RATE  = 16,
  HORIZ_POS_VERT_THR_YAW_ANG   = 17,
  HORIZ_POS_VERT_THR_YAW_RATE  = 18,
  GPS_ATII_CTRL_CL_YAW_RATE    = 97,
  GPS_ATTI_CTRL_YAW_RATE       = 98,
  ATTI_CTRL_YAW_RATE           = 99,
  ATTI_CTRL_STOP               = 100
}
```

actually it is the mode you set when doing attitude control

trickyMan, 2016-12-14:
Ok，then , It not  the < Flight Mode > in the iOS SDK. Could you please tell me how to get the corresponding < Flight Mode > in the Onboard SDK?

Here is the definition in iOS SDK:
[DJIFlightControllerFlightMode](https://developer.dji.com/iframe/mobile-sdk-doc/ios/Enums/DJIFlightControllerFlightMode.html)

Thank you!!!

lanyusea, 2016-12-14:
should be the `Flight status` as I remember

lanyusea, 2016-12-14:
seems not possible in current SDK version.

trickyMan, 2016-12-14:
(MAKE_VERSION(3, 1, 10, 0));
My Onboard Version is above.

lanyusea, 2016-12-14:
yes I know.

not possible in SDK v3.1, will be added in the next version

trickyMan, 2016-12-14:
OK，Thanks for you patients.

closed 2017-03-10 by lanyusea