dji-sdk/Onboard-SDK #84: SDK v3.2 fails CMAKE with error "Target dependency djiosdk-wrapper not found"

taurean1192, 2016-12-30:
The instructions on the readme page, lead to an error in compilation via CMAKE - Target dependency not found

zliDJI, 2017-01-03:
Hi, taurean1192. What onboard computer and OS version are you using?

taurean1192, 2017-01-04:
I am using a Raspberry Pi 3 (Arm 64 bit), running the Raspbian Jessie.

I also tried using Ubuntu 16.04 Arm version but it failed.

I also did the same with my x64 laptop running Ubuntu 16.04 but it failed
there as well.

wellsa1, 2017-01-04:
I have the same issue on Raspbian Jessie.  Haven't found a work-around yet.

zliDJI, 2017-01-04:
Please try to cd to the top level Onboard-SDK, and then
mkdir build
cd build
cmake ..
make

taurean1192, 2017-01-04:
I did that as well. The executable is then created in 'build/bin'

On running it, after pasting a copy of UserConfig.txt, it doesn't recognise
the mode LB2 is in and gives the error that RC is not in mode P, something
we observed in v3.1.9

zliDJI, 2017-01-04:
Hi, taurean1192, could you post a picture of your error?

Also, I assume you are running the latest firmware, and the RC is in F mode, right?

wellsa1, 2017-01-04:
This will only work on Ubuntu right?  I'm failing the cmake since it can't find a distro version that matches the *.cmake 16.04, 14.04 or 18.

taurean1192, 2017-01-04:
No I had the RC in P mode for the latest firmware. The instructions said to keep the it in P mode for FW 1.5 in A3. 
[image]

zliDJI, 2017-01-04:
Hi @taurean1192, I tried to reproduce your problem but was not able to. Please check in DJI GO if your firmware matches mine: Aircraft 1.3.0.00+ (it's actually 1.5 from dji assistant 2),  RC (1.1.70), LB2Air(1.1.2.00).

At the same time, could you delete your above post  and start a new issue, since this is a different problem.

I assume that by running cmake at top level, you don't have "Target dependency djiosdk-wrapper not found" issue any more, right?

Thanks!

taurean1192, 2017-01-05:
Hi @zliDJI I have the A3 (FW 1.5), LB2 Air (v1.1.4) and LB2 RC at (latest fw). You think this could be a possible issue?

taurean1192, 2017-01-05:
Also when compiling the v3.2 SDK over 14.04 , on doing CMAKE over the top level, it throws the following error - 

`if given arguements:

"GTEST" "OR" "(" "USE_PRECISION_MISSIONS" "AND" "(" "STREQUAL" "1604" ")" ")"

Unknown arguments specified

-- Configuring incomplete, errors occurred!`

taurean1192, 2017-01-05:
I just tried it on Ubuntu 14.04 (ARM 32 bit), the CMAKE inside sample/Linux/Blocking/build works without errors, but make throws the following error - 

`
[image]
`

taurean1192, 2017-01-05:
I am adding a screenshot of the latest errors, please see this
[image]

zliDJI, 2017-01-05:
Hi, @taurean1192 , with x86-64 ubuntu 1604, please run cmake at the top level build folder instead of sample/Linux/Blocking/build  

On manifold ubuntu 14.04, we just fixed a cmake bug and is currently testing. Will push in a couple of hours. Sorry about that.

zliDJI, 2017-01-06:
@taurean1192  We have pushed some cmake changes. Could you try to build to see if they solved your problem?

@wellsa1 You should be able to build (without advanced features) on Pi. Please let us known. Thanks!

taurean1192, 2017-01-07:
@zliDJI these are snapshots for a RPi 3 (64 bit ARM) running Raspbian Jessie (Debian 8.6). I tried build both at top level and at the 'sample/Linux/Blocking/build' levels
[image]
[image]

Then I treid doing this on Ubuntu 16.04 running on RPi 3, the following error comes - the same comes on my x64 Ubuntu machine. I updated A3 to latest FW v1.6, DJI Assistant 2 v1.0.9 but nothing changed. Is it possible that my LB2 is faulty

[image]

wellsa1, 2017-01-10:
I have the same errors as taurean attempting cmake with the Non-Blocking version and at the root.  Rasberry pi Jessie ARM 64.  It appears nothing changed.

zliDJI, 2017-01-12:
We pushed a branch "raspberrypi-support", and we have tested to build linux sample on raspberrypi 2. Please check it out and give it a shot. Thanks!

rohitsantdji, 2017-01-23:
Closing this issue. If errors persist, please reopen this issue or start a new one depending on the nature of the error.

closed 2017-01-23 by rohitsantdji