dji-sdk/Onboard-SDK #86: Data transparent transmission from mobile received incorrectly

alexandrubichis, 2017-01-12:
We implement an application on an OES device using OnboardSDK.
It receives some data from an android device running MobileSdk through Data Transparent transmission.
With A3 fw version 1.3.0.00 the OES receives the data correctly.

When changing to A3 fw version 1.6.0.10 with the same code the data is received to OES, but it can't be recognized.
The header of the DTT packet is the same, but the payload is nothing usable.

Here is an example of 3 bytes data packet received from MobileSDK:

A3 1.3.0.00:
aa 15 00 00 00 00 00 00 20 00 11 de 02 02 01 01 00 81 30 1f 5b

A3 fw 1.6.0.10:
aa 16 00 00 00 00 00 00 b6 81 aa ee 02 02 20 73 44 20 af aa 9a b7

Could it be that the data is transmitted encrypted on the new fw 1.6.0.10?

Note:
1.	While changing code to work on fw 1.6 we updated to the newest OnboardSDK 3.2 and MobileSDK for Android 3.5.
 The A3 with fw version 1.3 receives the data correctly.
 
2.  When using A3 1.3.0.00 the activation of OnboardSDK is requested with 
  "versionA3_31 = (MAKE_VERSION(3, 1, 100, 0));" (successfully),
 while when using A3 1.6.0.10 the activation of OnboardSDK is requested with: 
 "versionA3_32 = (MAKE_VERSION(3, 2, 100, 0));" (also successfully).

lanyusea, 2017-01-12:
The release of 1.6 firmware is a mistake, two major fixes related to SDK are not merged.
1. the fix of transparent transmission
2. the fix of sdk version

Sorry about the inconvenience, we team run into some misunderstandings when doing version planning on a branch between 1.3 and 1.6.

There will be a hot fix on or before Jan.20 to solve all these(more than these two, actually) issues.

It's our mistake, sorry about it.

alexandrubichis, 2017-01-12:
Thank you for quick reply. We have same issues on M600?

lanyusea, 2017-01-12:
yes

rohitsantdji, 2017-01-23:
New FW 1.7.0.0 (released 01/18) for the A3 should have resolved these issues. Please reopen this issue if you continue to see errors.

closed 2017-01-23 by rohitsantdji

alexandrubichis, 2017-02-07:
Is this issue fixed on M600 in latest relases V 1.0.0.80(2017.01.05)? 

Thank you

rohitsantdji, 2017-02-07:
Sorry, it is not fixed in V1.0.0.80. The next M600 release should fix this.

alexandrubichis, 2017-02-07:
Thank you for your quick reply. Do you know when is planned the next release?

alexandrubichis, 2017-02-21:
Do you have an estimate date when the new M600 will be release?