dji-sdk/Onboard-SDK #87: N3 Support

a2hill, 2017-01-18:
It is unclear to me what the status of N3 support is for the OSDK v.3.2.0. On their website DJI says that the N3 is compatible with the MSDK and OSDK, however there is no version constant in DJI_Version.h to for the N3 and it does not seem to be mentioned in the release notes. In my testing I've been using `versionA3_32` and am able to activate, however I am running into some issues. Is the N3 a supported OSDK platform?

lanyusea, 2017-01-18:
yes, and should be the same as A3.

what are the issues?

a2hill, 2017-01-18:
I can't get Virtual RC control enabled. The same code running on an M100
with OSDK 3.2 works fine, but using the N3 CtrlInfo never shows open for
virtual RC. Additionally when I query the version from VersionInfo the
value does not match any of the constants in DJI_Version.h

lanyusea, 2017-01-18:
The VRC has been removed from our feature list, both A3 and N3.
M100 has no more ROM for new firmware so we didn't give it a shot.

Yes, there is a version query bug in both A3 and N3, will be fixed in next version.

a2hill, 2017-01-18:
That's really a shame about the VRC functionality, have been using that on the m100 for some projects and it seemed promising. 

About the m100 it being out of ROM: does this mean that it is end of life and we should be expecting a new development platform soon? I would assume it being incompatible with the latest version of the OSDK would indicate that. I would feel a little silly buying more units knowing that it has a limited future.

Additional issue: When using the N3 and obtaining control via `setControl` I am able to obtain control, however I get a repeated error in my console saying: 

> ERROR recvReqData,line 217: Unknown mission code 0x92

I don't see this ID anywhere in the documentation, is this a bug?

lanyusea, 2017-01-19:
Hi a2,

1. In our original design, VRC is for a second controller implementation
besides the DJI-provided LB2.
while, we found most of people use it to let the drone not RTH while RC
signal lost.
So we removed this feature and add an option in DJI Assistant 2 instead.

2. It's not end-of-life, it is somehow like Win10 software are not
compatible with Win7 but Win7 has its own software support. we will keep
maintaining the M100 but from software part, there should be no new
features merged in in the future, while bug fix still yes, SDK support
still yes, after-sale still yes. Currently, to stay with our latest SDK
feature, A3 and N3 are the only options.
I cannot say if there will be a new platform in the future.

3. I think you need to upgrade your N3 firmware.

a2hill, 2017-01-19:
1. Very good, I was wondering what that option in Assistant 2 was. Thanks for clarifying.

2. I understand, thank you.

3. According to Assistant 2 the firmware is at the latest version. I think there was a firmware update that was pulled due to a bug?

rohitsantdji, 2017-01-23:
3. New FW 1.7.0.0 (Released 01/18) is stable, please upgrade to that. If you still see issues, please reopen this issue or start a new one depending on the nature of the issue. Thanks!

closed 2017-01-23 by rohitsantdji