dji-sdk/Onboard-SDK #88: 'Failed to obtain control' in Onboard SDK v3.2

taurean1192, 2017-01-24:
In continuation of similar issues I have faced previously, the SDK's sample has failed to obtain control of my DJI A3 FC. It flashes the same error - RC not in P mode, while my RC is in single flight mode - P.

The compilation raises no error. I am running the A3 FW 1.7. The system is a Raspberry Pi 3 (ARM 64bit) running Ubuntu 16.04.
[image]

rohitsantdji, 2017-01-24:
For non-x86 systems, a bug in osdk-wrapper incorrectly shows successful activation while the system may not actually be activated; the sample might have failed at any point before attempting to obtaining control. This bug will be fixed in an upcoming release, but till then you cannot rule out the possibility that the connections are incorrect, the baud rate does not match, the buffering on your USB-serial adapter is not correctly configured or activation has not gone through. 

To help debug - could you try the same sample on an x86 system with the same connections, and post your findings here?

taurean1192, 2017-01-24:
Hey @rohitsantdji I had thought of the other possible errors as well hence I checked all of that and ran the sample with different baud rates, loose connections, checking serial buffer and the others. All of them checked out. This shows the same thing on my x64 laptop. So two platforms won't show the same problems after so much verification.

Also this issue is persistent since v3.18 of the SDK, then I tried doing it on the x86 ARM, but the same issues prevailed. Now it is difficult to find the 16.04 OS distro for the x86 version of ARM, the existing one often faces bugs upon updating.

rohitsantdji, 2017-01-24:
Okay. Output on a x64 laptop is also good enough for our debugging purposes.

rohitsantdji, 2017-01-24:
A screenshot of DJI Assistant 2's SDK page and the default page will also be helpful.

jvgomez, 2017-01-25:
For completeness, this issue seems to be a general problem. I am facing the same issue as well (I cannot make the demo code work in a x64 PC), but I have seen some other posts with the same/similar problems:

http://forum.dev.dji.com/thread-33009-1-1.html
http://forum.dji.com/thread-81078-1-1.html

taurean1192, 2017-01-25:
@rohitsantdji please see below the output for the x86 ARM machine, running Ubuntu 16.04. It gives the same error. 
[image]

Also as you requested the SDK settings using the DJI Assistant 2 software
[image]

rohitsantdji, 2017-01-25:
@jvgomez Thanks for the additional links.

@taurean1192 it seems from your screenshot that this is still an ARM computer. (by [x86 ARM](http://stackoverflow.com/questions/14794460/how-does-the-arm-architecture-differ-from-x86) maybe you meant 32-bit ARM?). If you get a chance, please try it on Ubuntu 16.04 on an Intel/AMD x86_64 computer and let us know if you are getting a "Your SDK Version in User_config.h does not match the one reported by the drone." error.

To give everyone an update on this: FW 1.7.0.0 introduces new version strings for N3 and for A3, and the SDK currently does not have support for this. This causes activation itself to fail, and a good way to check this is to run the sample on an x86 machine where the serial driver can implement various levels of checks. Expect a hotfix soon.

jvgomez, 2017-02-01:
@rohitsantdji is there any expected date for the hotfix? This is blocking some other things in my case. Thank you.

FLDataTeK, 2017-02-01:
Agreed we need to know when to expect the hot fix. This issue has halted our development and we need to get things moving.  We are going to load up a x86 board this morning so we can make some progress but we need a solution to this issue ASAP so we can continue down the proper path as we have already wasted to much time on this.

rohitsantdji, 2017-02-01:
Hotfix is in testing and will be released by tomorrow (02/02). Apologize for the delay, guys - our tests discovered some latent issues when run on Ubuntu 14.04 so we're fixing those.

FLDataTeK, 2017-02-01:
We tried this on a x86 (Intel Atom) based machine and seem to be getting further along, but we are unable to activate the SDK. 

x86 Intel Atom
Ubuntu 14.04
USB to Serial cable going to the API port on the A3
The PC has internet connectivity
Using the current versionA3_32 = (MAKE_VERSION(3, 2, 15, 37)); in DJI_Version.h  as provided by DJI dev support

FrSky Taranis Plus with SBUS receiver connected and turned on and showing in P mode in Assistant 2

It does look like it is talking to the FC now, but we are unable to activate. We have ran a make & make clean 

[image]

rohitsantdji, 2017-02-01:
@FLDataTeK you'll need a LightBridge2 RC with DJI GO connected to the internet to activate the device.

FLDataTeK, 2017-02-01:
We were  told the last SDK and Assistant 2 update removed this requirement for LB2.

rohitsantdji, 2017-02-01:
If anyone (@taurean1192 @jvgomez ) is blocked by this, please follow the support suggestion to @FLDataTeK   (set versionA3_32 = (MAKE_VERSION(3, 2, 15, 37)); in DJI_Version.h); the fix implements a new automated versioning system that won't require you to find our what version your flight controller is running.

rohitsantdji, 2017-02-01:
@FLDataTeK does the program return the same error when run multiple times? And are you running Assistant 2 v1.0.9-2?

FLDataTeK, 2017-02-01:
@rohitsantdji 

Yes we have ran multiple times and are running the latest firmware rev and Assistant v1.0.9-2

rohitsantdji, 2017-02-01:
@FLDataTeK The current implementation of the non-LB2 activation can activate without an LB2 RC, but still requires an LB2 air system connected to the A3. This requirement will be removed in the next version.

jvgomez, 2017-02-01:
@rohitsantdji thanks for the update!

natelannan-osu, 2017-02-02:
I still got a mismatch error using versionA3_32 = (MAKE_VERSION(3, 2, 15, 37));
I'm running ubuntu 14.04 on an ARM BeagleBone Black

just fyi for others running similar HW

rohitsantdji, 2017-02-03:
Please try the 3.2.1 release and report any issues. Thanks!

natelannan-osu, 2017-02-03:
@rohitsantdji 
I don't think parseDroneVersionInfo() is working correctly for me.  I am getting null characters in buf after memcpy(buf, ackPtrIncoming, 64);.  Wondering if something is going screwy with ptemp in getDroneVersion().  going to look into it further, but if you have suggestions that would be grand.

HW: M600
FW: V1.0.0.80
Assistant: V1.0.9-2
SDK: 3.2.1
Onboard device:  ARM BeagleBone Black running Ubuntu 14.04

output (I added some debug statements):
root@arm:~/newSDK/Onboard-SDK/build/bin# ./djiosdk-linux-sample -interactive
User Configuration read successfully.

These are your User_Config settings.
Serial port = /dev/ttyO4
Baudrate = 230400
App ID: *****************************
App Key: ***************************

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 51: Attempting to open device /dev/ttyO4 with baudrate 230400...
STATUS init,line 60: ...Serial started successfully.
before getDroneVersion(1)
parseDroneVersionInfo:
buf[0]: 0x00
ackPtrIncoming + 0: 0x00
buf[1]: 0x00
ackPtrIncoming + 1: 0x00
buf[2]: 0x00
ackPtrIncoming + 2: 0x00
buf[3]: 0x00
ackPtrIncoming + 3: 0x00
buf[4]: 0x00
ackPtrIncoming + 4: 0x00
buf[5]: 0x00
ackPtrIncoming + 5: 0x00
buf[6]: 0x00
ackPtrIncoming + 6: 0x00
buf[7]: 0x00
ackPtrIncoming + 7: 0x00
buf[8]: 0x00
ackPtrIncoming + 8: 0x00
buf[9]: 0x00
ackPtrIncoming + 9: 0x00
buf[10]: 0x00
ackPtrIncoming + 10: 0x00
buf[11]: 0x00
ackPtrIncoming + 11: 0x00
buf[12]: 0x00
ackPtrIncoming + 12: 0x00
buf[13]: 0x00
ackPtrIncoming + 13: 0x00
buf[14]: 0x00
ackPtrIncoming + 14: 0x00
buf[15]: 0x00
ackPtrIncoming + 15: 0x00
buf[16]: 0x00

In the previous SDK this info was coming down in getDroneVersion():
my debug from 3.2.0:
version_name:  v1.0 BETA PM820V3-03.02.10.00
version_ack:  65281
version_crc:  808924720
version_ID:  -1094574206
version:  50462720

thanks for your time

rohitsantdji, 2017-02-03:
Could you uncomment the print block at the end of parseDroneVersionInfo() and report the output?

natelannan-osu, 2017-02-03:
It never makes it that far.  it jumps out while parsing the name:

  while (*ackPtr != ' ') {
    ackPtr++;
    if (ackPtr - startPtr > 64) {
      API_LOG(serialDevice, ERROR_LOG, "Drone version was not obtained. Please restart the program or call getDroneVersion\n");
      return false;
    }
} //! Found first space ("SDK-v1.x")

I can take out those returns and see what we get if you want

rohitsantdji, 2017-02-03:
At first glance this looks like a partly unrelated serial driver issue - we do not test on BeagleBone, so it is difficult to debug this offline. On ARM devices the serial driver does not validate the serial connection, so zeros on the line could be caused by any layer of the serial stack. Results of a test from an x86 computer or a Manifold would be helpful.

Is OSDK 3.2.0 returning useful information with this setup as it is right now?

natelannan-osu, 2017-02-03:
It was yielding useful data yesterday.  The only thing I did today was clone 3.2.1, add my user_config.h and build.  Drone stayed unchanged.  I am away from my work station right now so I will have to revert to 3.2.0 when u get back

rohitsantdji, 2017-02-04:
A small fix has been issued to address a bug in M600 and M600 Pro aircraft running FW 1.0.0.80, please try that if you own one of these aircraft.

rohitsantdji, 2017-02-07:
Fixed in OSDK 3.2.1. Please re-open if this is not the case for you.

closed 2017-02-07 by rohitsantdji

MonkaSebastian, 2017-02-10:
Hey,
last week there wasn't any problem with the onboard sdk but now there is.
The last time when it worked I used the OSDK 3.2 but suddenly it failed to obtain control.
(Do I need to activate through the internet again, when I change and compile my program again?)

I tried so.
I changed my SDK to 3.2.2 deleted my version: versionM100_31.
But now I get this error with the native program.

root@EMC2_LVTTL:/mnt# ./EMC2_Matrice4.elf -interactive
User Configuration read successfully. 

These are your User_Config settings.
Serial port = /dev/ttyUL1
Baudrate = 230400
App ID: ......
App Key: .....

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 51: Attempting to open device /dev/ttyUL1 with baudrate 230400...
STATUS init,line 60: ...Serial started successfully.
ERROR parseDroneVersionInfo,line 246: Drone version was not obtained. Please restart the program or call getDroneVersion

Attempting activation..
ERROR parseDroneVersionInfo,line 246: Drone version was not obtained. Please restart the program or call getDroneVersion
Automatic activation successful.
Failed to obtain control.
Your RC mode switch is not in mode F. (Is the RC connected and paired?)
This program will exit now.

taurean1192, 2017-02-13:
I tried with SDK v3.2.1 today, it still isn't working properly on mine. I am using ARM 32 bit, Ubuntu 16.04.
The output is as follows - 

pi@pi-desktop:~sudo ./djiosdk-linux-sample -interactive
User Configuration read successfully.

These are your User_Config settings.
Serial port = /dev/ttyUSB0
Baudrate = 115200
App ID: ......
App Key: .....

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 51: Attempting to open device /dev/ttyUSB0 with baudrate 115200...
STATUS init,line 60: ...Serial started successfully.
ERROR parseDroneVersionInfo,line 246: Drone version was not obtained. Please restart the program or call getDroneVersion

Attempting activation..
ERROR parseDroneVersionInfo,line 246: Drone version was not obtained. Please restart the program or call getDroneVersion
Automatic activation successful.
Failed to obtain control.
Your RC mode switch is not in mode F. (Is the RC connected and paired?)
This program will exit now.

What to do? 3 months since purchasing but still stuck at the same point

rohitsantdji, 2017-02-13:
@taurean1192 based on the response of other users and community members it appears that the issue has in general been resolved. It sounds like the issue you are seeing has something to do with the particular setup that you have. Raspberry Pi platform support is still experimental - we can help you debug this. Are you using a Pi 2?

taurean1192, 2017-02-14:
@rohitsantdji yes I am using Pi 2, I also have Pi 3 with me.

taurean1192, 2017-02-17:
@rohitsantdji I tried testing the Raspberry-support version of the SDK on Pi2, it gave the following error.

[image]

Please advise the way forward

taurean1192, 2017-02-17:
@MonkaSebastian  what is your OES?

MonkaSebastian, 2017-02-17:
Hey I use "Petalinux", it is a Yocto based linux for Xilinx FPGAs.
The cpu is also arm 32.
Its a really random error. Proof your UART connection and your connection between FC, controller and internet.
Good luck!

taurean1192, 2017-02-17:
@MonkaSebastian thanks. I am looking for a SoC to run this setup of A3 on. Spent months trying on RPi. Any SoC you can suggest? @rohitsantdji  can you suggest any SoC running Linux?

FLDataTeK, 2017-02-21:
@taurean1192 We have started using Intel Atom based Minnowboard with success. We have successfully test flown using the SDK demos and now are working on getting our software moved over to it and flight tested.

taurean1192, 2017-03-01:
@FLDataTeK any particular version of the MinnowBoard and the OS you are using that you can tell me?

adamwathieu, 2017-07-11:
Hey,
Did anybody figure out how to fix this issue. I have some deadlines, and I need to get this fixed so I can move on to the real stuff:
https://github.com/dji-sdk/Onboard-SDK/issues/143
Thanks.