dji-sdk/Onboard-SDK #90: "Your SDK version in User_Config.h does not match the one reported by the drone"

jrmzelda, 2017-01-26:
I am trying to run the "DJI Onboard SDK C++ Linux Example". I have the raspberry-support branch, and I followed the CMake build steps for 3.2.0. After completing the build and running djiosdk-linux-sample -interactive I get the following error output:

...
STATUS init, line 60: ...Serial started successfully.

Attempting activation..
Your activation did not go through.
Your SDK version in User_Config.h does not match the one reported by the drone. 
Please correct that, rebuild (do a 'make clean' first) and try again.

The SDK Version is shown as 3.1, my User_Config.h and UserConfig.txt both have versionM100_31, and my ID and KEY are input correctly.

I have also tried to follow the build steps for version 3.1.9. When I follow those steps, I run into more issues. Notably, upon make'ing the project I get these lines:
"/usr/bin/ld: cannot find -ldjiosdk-core
/usr/bin/ld: cannot find -ldjiosdk-platform
/usr/bin/ld: cannot find -ldjiosdk-wrapper"

Among other errors...

Should I abandon ship and install ROS instead?

rohitsantdji, 2017-01-26:
Is your OES a Pi 3 or a Pi 2? Also what FW is your M100 running?

jrmzelda, 2017-01-26:
Pi 3. I will check M100 FW tomorrow, but I believe it is the most recent (I used DJI Assistant 2 to update today).

rohitsantdji, 2017-01-26:
Another thing you should try: in the file osdk-wrapper/src/LinuxSetup.cpp, uncomment the call to api->setVersion() on line 201.

jrmzelda, 2017-01-26:
I uncommented line 201 as you suggested, and I no longer get the original error message. 

By the way, here is some of the output now that it successfully ran:
"STATUS getDroneVersionCallback, line 599: version ack = 0
STATUS getDroneVersionCallback, line 602: version crc = 0xA6453AAC
STATUS getDroneVersionCallback, line 599: version ack = 0x41DD92920
STATUS getDroneVersionCallback, line 599: version name = SDK - v1.0 BETA M100-03.01.01.00"

Thank you for the help!

closed 2017-01-26 by jrmzelda

YuyangGeng, 2017-03-17:
I'm using PI 3 and the raspberry-support branch, following the CMake step for 3.2.0 too. After finishing "cmake .." command, nothing generates in "build/bin" folder. Did I miss something important?

rohitsantdji, 2017-04-06:
@YuyangGeng  - type the `make djiosdk-linux-sample` command after the CMake step.