dji-sdk/Onboard-SDK #92: moveByPositionOffset limitations?

jrmzelda, 2017-02-03:
Hi,

I am editing the Linux Interactive Sample to allow for up, down, left, right, forward, backward, and yaw commands. I am using commands like "moveByPositionOffset(api, flight, 0, distance, 0, 0)" to do so. I have run into a few issues. I am using DJISimulator to inspect the results.

1. When the distance is more than ~50 meters, the drone certainly does not fly the full distance. I have tried to make a 100 x 100 square, but the drone does not fly the full 100 meters. Is there a limit to how long the drone can fly before control is passed back to the program?

2. When I send yaw commands (90 degrees left or right), the drone sometimes responds and sometimes does not. Eventually, if it does not respond to a few commands, it seems to accumulate the past commands and execute them all at once. For example, I command it to turn 90 degrees left twice with no response. On the third try, the drone yaws left 270 degrees. Is there a reason these commands are buffered and unacknowledged frequently?

3. On an unrelated note, can I use the OES to specify RTH altitude with a specific function? On a per-flight basis, I would like to implement a feature to do so. I know that DJI Go has this feature, but I am doing all of my flight control from a a laptop instead.

rohitsantdji, 2017-02-04:
1. `int moveByPositionOffset(CoreAPI* api, Flight* flight, float32_t xOffsetDesired, float32_t yOffsetDesired, float32_t zOffsetDesired, float32_t yawDesired ,  int timeoutInMs = 10000, float yawThresholdInDeg = 1, float posThresholdInCm = 5);` If you notice the declaration in LinuxFlight.h, there are some optional arguments with default values, timeout being one of them; i think you are reaching the timeout in this case. Try setting the timeout to a larger value.

2. This does not sound like expected behavior. Could you provide more information with maybe an example of the calls you are making? 

3. Sorry, the RTH altitude cannot be configured through the Onboard SDK. DJI Go and the Mobile SDK use a different communication protocol and have access to an extended set of commands the the OSDK does not.

jrmzelda, 2017-02-04:
Thank you for your responses! I suspect timeout is my issue as well. 

Here is my yaw control code for a 90 degree right yaw: 
int currentYaw = 0;
currentYaw += 90
if(currentYaw == 360)
  currentYaw = 0;
positionControlByOffset(api, flight, 0,0,0,currentYaw);

I have the if to keep the values between (-360 and 360) in case that is an issue. I also am using int, and just realized this calls for float32_t. But I do not think those are part of the problem since it works sometimes. It could be unreliable serial communication I suppose (I have a laptop ssh'd to the Pi, which is serially connected to the Matrice 100).

closed 2017-03-13 by rohitsantdji