dji-sdk/Onboard-SDK #94: Waypoint Mission Crashing DJI GO app

natelannan-osu, 2017-02-08:
I logged an issue to DJI as well since I suspect the problem is in the android DJI Go app.  I will update here as I hear back from support.  Running a Waypoint mission that ran successfully using previous SDK version is now crashing android DJI Go app.  The aircraft continues to run mission after app has crashed.

M600 FW - 1.0.0.80
android DJI GO version - 3.1.2
OSDK version -  3.2.1 with M600 patch

ios app 3.1.4 does not crash

natelannan-osu, 2017-02-19:
DJI dev support got back to me.  This is a known issue they are actively working on.

closed 2017-02-19 by natelannan-osu

chukcha2, 2017-02-24:
why close it if it is not fixed yet or without referencing another issue that tracks it?