dji-sdk/Onboard-SDK #96: The returned huge velocity is not conform to movement control command that I sent to the STM32

wangweiboss, 2017-02-13:
Hello,
    I controlled the movement of  M100 by using Onboard SDK STM32 sample in a simulator. Unfortunately, the returned huge velocity in Realterm is not conform to movement control command that I sent to the STM32, for example, I sent a commend  '0xFA 0xFB 0x04 0x01 0x41 0x00 0x00 0x80 0x0A 0x00 0x00 0x00 0x00 0xFE' which means the velocity of Y axis is -0.1m/s, whereas the real velocity of  Y axis is -322.5m/s, which leading to an accident. 
   I think that the error is mainly due to  that I sent the command too frequently, I scan the main.cpp, Is there any possible that "sendPoll()" function exists any bugs?
 I am looking forward to your reply.

Sincerely,
Kevin  
[image]

zliDJI, 2017-02-13:
Hi, are you running the sample code with your own number, or you are writing your own code?  Does the command in the doc work (0xFA 0xFB 0x04 0x01 0x48 0x00 0x64 0x00 0xC8 0x00 0x64 0x00 0x05 0xFE)?

What is your set up, in terms of 
1. flight controller FW version
2. sdk version (github commit id)
3. frequency settings in your code (you said you suspect you were sending too frequently)
4. frequency settings on dji assistant 2

Thanks!

wangweiboss, 2017-02-14:
Hi, aliDJI

Thanks for replying. Here are my answers:
I am running the sample code with my own number. this command worked well when I  sent the command not that frequently.
1. the FW version is N1,V1.3.1.00
2. the SDK version is "onboard-SDK-3.2.0"
3.  the frequency I sending is approximately 5s interval per command.
[image]

4.  the frequency settings on dji assistant 2 are showed in picture below. these are all in default mode.

wangweiboss, 2017-02-14:
N.B Messages that  returned to Realterm sometimes mentioned that "Error interface ,there is not enough memory"

zliDJI, 2017-02-15:
> the frequency I sending is approximately 5s interval per command

How did you send the same command at 1/5=0.2Hz? Using some auto send function by realterm?

In the sample code, you are not supposed to send the movement command multiple times. Inside the code, when you send 1 movement command from realterm, the firmware code sets up a 50Hz timer and the timer interrupt resends the command.

wangweiboss, 2017-02-15:
Hi, aliDJI
Thanks for replying. Here are my answers:
       Yeah, I send the command using auto send function by realterm. And when I click the send menu on the relaterm software panel **by hand**, this circumstance/error **still occur**.
I looked into the code in the sample stm32 project. Inside the code, the timer interrupt would resend the command, just like you said. But I need to control M100(4 directions:left, right, forward, backward), I couldn't send the command just once. So what mean of "you are not supposed to send the movement command multiple times".
Thanks for your reply!
Sincerely.

zliDJI, 2017-02-15:
In order to achieve what you want, one way is to write your own loop that periodically (<50Hz) send movement control commands to M100. There is an example for STM32 called Local Navigation which moves the drone to follow a rose shaped curve, please see the documentation.

If you just want to use the sample FW as your tool without writing your own code,  another possibly easier way you can try is to comment the line `TIM_Cmd(TIM2, ENABLE);` in Receive.cpp.  In this case, when you send a command in real term, that command will be just sent once, timer and resending won't be started. Then you can use your auto send function in realterm to send your command (e.g., left for 1s, right for 1s, forward for 1s...) at around 20Hz.

wangweiboss, 2017-02-15:
Hi, aliDJI，
Thank you so much for your advise, I just tried to comment the line TIM_Cmd(TIM2, ENABLE) in Receive.cpp, but M100 couldn't  move left, right, up and down, and it can also take off and landing. I think  TIM2 is not responsible for the issue i mentioned above. Is there any other ways to solve it?

Thanks again.

zliDJI, 2017-02-17:
Hi. We tried several times and we could not reproduce your problem of huge velocity. 

Can you please post a screen shot when the huge velocity happens? We need screen shot of:
1. The DJI simulator window.
2. The realterm  window with "send" tab.

Thank you!

rohitsantdji, 2017-04-06:
Closing since there has been no activity for a while, and we cannot reproduce the error.

closed 2017-04-06 by rohitsantdji