dji-sdk/Onboard-SDK #98: Data transparent transmission not working with M600

alexandrubichis, 2017-03-02:
Using latest firmware V1.0.0.80(2017.01.05), data transparent transmission not working. This issue is related with [https://github.com/dji-sdk/Onboard-SDK/issues/86](url).

Do you have an estimate date when the new M600 will be release with this fix?

rohitsantdji, 2017-03-06:
A new M600 FW will be out by the end of the month, with a fix for this issue. We apologize for the delay.

tudorDME, 2017-04-27:
Any word on when an updated firmware will be out that supports this? The M600 Pro cannot be downgraded below this firmware which basically makes it unusable with the onboard SDK.

tudorDME, 2017-05-10:
Everything works now with the new firmware that was released 5/2/2017 works (v1.0.1.20).

closed 2017-05-11 by alexandrubichis