dji-sdk/Onboard-SDK #99: OSDK activation gets stuck midway in Linux

taurean1192, 2017-03-07:
I am testing the setup on a A3 +LB2, and my Laptop (Intel i5) (as OES) running Ubuntu 16.04. The activation process gets stuck, see the image below
[image]

I attached the UART RXD pin to TX of my USB to TTL, and UART TXD pin to RX of the TTL. Since there was no response coming on the screen I assumed maybe the connection was faulty hence I even reversed the connection, but nothing happened.

Some help will really be appreciated.

lanyusea, 2017-03-07:
any print when `cat /dev/ttyUSB0` ?

taurean1192, 2017-03-07:
@lanyusea there is no print, just blank when try to cat the serial stream.

EduardoCordeiro, 2017-03-07:
I had the exact same problem.

We mistakenly connected the TX on OES to the TX on Matrice. The Arquitecture guide shows it should be TX <-> RX; RX <-> TX.

Check if this solves it.

lanyusea, 2017-03-07:
if there is nothing it means either the hardware connection is not okay, or
you forgot to enable SDK in DJI Assistant

taurean1192, 2017-03-07:
@lanyusea I had checked for the SDK enable before starting it. Could it be a fault with the TTL converter?? 

@EduardoCordeiro what OES did you use exactly?

EduardoCordeiro, 2017-03-07:
I used a Ubuntu 16.04 Laptop, with a second windows computer connected using the DJI assistant.

lanyusea, 2017-03-07:
maybe, if there were a connection, you should be able to read the data sent
from the drone by `cat` the device

taurean1192, 2017-03-07:
@EduardoCordeiro that is setup I am using too, what baud rate?

@lanyusea I get that but the screen was showing blank. The TTL converter had it's 'blue data led' continuously on.

EduardoCordeiro, 2017-03-07:
I used the default one in the UserConfig.txt file.
Are you using the New Command Line example to activate? 
You need to make a copy of this file after you edit it [ID and Key].
Also you need this command: sudo usermod -a -G dialout $USER and then logout and login again.

taurean1192, 2017-03-07:
@EduardoCordeiro I did all of that. I even used picocom to listen to any data coming from the FC at 115200 baud rate, but it is showing as if it is reading nothing

EduardoCordeiro, 2017-03-07:
What I did was:
cmake and make on the Blocking example as instructed;
Edit the UserConfig file to match the ID and Key given by DJI
connect the M100 to the windows pc and enable SDK control, open Simulator
connect the M100 though USB to the OES (Ubuntu 16.04)
turn on radio controller (make sure it is paired)
turn on DJI Go App on the Camera screen
run the usermod command, logout, login again
./djisdk-sample-blocking

My problem was having the wires not well connected.

taurean1192, 2017-03-07:
@EduardoCordeiro I am working using the 3.2.1 version of the OSDK, the instructions now say to build from the top level itself. So I did that and got the executable built.
I copied the UserConfig.txt and edited it
I connected my A3 through a USB-TTL converter
Got my LB2 connected to a smartphone and the Go app
did the usermod, logout , login....

Now I am stuck

EduardoCordeiro, 2017-03-07:
You also need a secondary PC connected to the DJI Assistant I think.
But if the USB TTL connection is correct, I believe you did the same as me

taurean1192, 2017-03-07:
@EduardoCordeiro I had that running, my Desktop running DJI Assistant 2

rohitsantdji, 2017-03-07:
Some more follow up questions: 

1. On the Assistant SDK page, is at least one item broadcasting at a non-zero frequency?
2. Do you get the same error when you change both Assistant and OES to 230400 baud?

taurean1192, 2017-03-16:
@rohitsantdji finally I was able to run it on a x64 based system as OES. Now I'll be looking at putting it on something like the minnowboard.

closed 2017-03-16 by taurean1192

jairoroke, 2017-04-25:
Hello friends      @EduardoCordeiro @taurean1192 @lanyusea    I am new to this activate the M100 in Linux but believe that they can help me if my connection is the correct way, I would not want to spoil the drone, Thanks friends, Regards! Here are the pictures! 

[https://1drv.ms/f/s!AiINUVmw6ZuTw3TJL6b8OhSuyEVa](url)

In the Linux configuration it should be, but I'm not sure if I'm right
[https://1drv.ms/i/s!AiINUVmw6ZuTw3ubihsFkzuXEhLs](url)

Thanks!