dji-sdk/Onboard-SDK #102: RPi branch sample indicates ".. is not in mode F" when the remote is in mode F

newtoniant, 2017-03-08:
Compiled the RPI support branch of the osdk, built the project, and ran the djiosdk-linux-sample program.  My remote is in F mode, but the SDK is saying that it is not in F-mode and it exits.  The sample code works on a linux PC I've been using for development, but when run on the RPi it fails because it thinks the mode is not F.

```
./djiosdk-linux-sample
User Configuration read successfully. 

These are your User_Config settings.
Serial port = /dev/ttyAMA0
Baudrate = 115200
Drone/FC Version: M100
SDK Version: 3.1
App ID: [removed]
App Key: [removed]

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 51: Attempting to open device /dev/ttyAMA0 with baudrate 115200...
STATUS init,line 60: ...Serial started successfully.

Attempting activation..
Automatic activation successful.
Failed to obtain control.
Your RC mode switch is not in mode F. (Is the RC connected and paired?)
This program will exit now. 
```

rohitsantdji, 2017-03-13:
From Issue #77 : 

The "mode F" error message is not completely reliable on ARM devices - sometimes, we have seen the program actually error out at an earlier stage (activation for example) but still proceed to the obtain control step where it errors out with this message. This is because the way the serial driver is implemented works differently with USB-Serial adapters than with native serial ports. A few possible causes:

Did you change the baud rate in both the SDK page on DJI Assistant 2 as well as in UserConfig.txt?
Are your connections correct?
Have you enabled API control?

taurean1192, 2017-03-14:
@newtoniant I have faced a similar issue with RasPi, trying to set on a new OES.. They need to work on their support to ARM based OES.

newtoniant, 2017-03-14:
@rohitsantdji here are my test results.

With a fresh git clone of the raspberrypi-support branch I used both a USB serial FTDI232 (ttypUSB0) and RPI GPIO serial (ttyS0).

Both serial methods gave the same results, and the results were different only based on the baud rate selected in UserConfig.txt and Assistant2 SDK settings.  As a note, I have successfully developed for a while on my desktop Linux.

With 230400 baud:
```
STATUS init,line 51: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS init,line 60: ...Serial started successfully.

Attempting activation..
Automatic activation successful.
Failed to obtain control.
Your RC mode switch is not in mode F. (Is the RC connected and paired?)
This program will exit now. 
```

With 115200 baud:
```
STATUS init,line 51: Attempting to open device /dev/ttyUSB0 with baudrate 115200...
STATUS init,line 60: ...Serial started successfully.

Attempting activation..
Your activation did not go through. 
Your SDK version in User_Config.h does not match the one reported by the drone. 
Please correct that, rebuild (do a `make clean` first) and try again.
This program will exit now. 
```

The raspberrypi-support branch has versionA3_31 as the version in UserConfig.txt, where the master branch has versionM100_31. I am using an M100, but tried both versions with no change in results.

newtoniant, 2017-03-14:
Just to clarify, the USB serial adapter that works well on my desktop is what I plugged into the RPi.  Working on the desktop, but not on the RPi.

newtoniant, 2017-03-16:
If anyone has gotten the OSDK to work on a Raspberry Pi, can you post the OS version you are using and the SDK version, whether it's the raspberrypi-support branch, or master branch.  Many thanks.

newtoniant, 2017-03-16:
@rohitsantdji @lanyusea

What embeded system are you using for the OSDK development?  What system that can fly (not desktop).  I don't have to use a Pi, but I certainly can't fly my desktop on an M100.

rohitsantdji, 2017-03-17:
@newtoniant : The raspberry-pi support branch (experimental) is only for Pi 2 on Raspbian, since the version of gcc it ships with does not support C++11 directly. If you have a Pi 3, you should give it a shot with the default 3.2 branch. 

Regarding supported OES, the OSDK is tested with Ubuntu 14.04 on Manifold (ARM, Jetson TK1), Ubuntu 14.04 and 16.04 on Core i5-based NUCs (NUC6i5SYK to be precise) and STM32F4 Discovery (bare metal, ARM Cortex M4).

newtoniant, 2017-03-17:
Confirmed that Pi 2 works if you use the raspberrypi-support branch, **AND**:

osdk-wrapper/src/LinuxSetup.cpp, uncomment the call to api->setVersion() on line 201.

As mentioned in [Issue #90 ](https://github.com/dji-sdk/Onboard-SDK/issues/90)

rohitsantdji, 2017-04-06:
Closing since we seem to have reached a good understanding of what works with Pi. If there are specific questions please open a new ticket.

closed 2017-04-12 by newtoniant