dji-sdk/Onboard-SDK #104: M100 + TX1,run Onboard-sdk linux sample  djiosdk-linux-sample on TX1, some error encoutered.

lijiangyong01, 2017-03-10:
M100 connect TX1(ubuntu 16.04 ,64bit) with usb serial,I test the usb serail with minicom, uart data can be received. But I run Onboard-sdk linux sample,there are some error.  everything is ok when M100 connect laptop (ubuntu 14.04 , 64bit) with usb serail.
M100 + TX1, error shows below:

[image]
I need some help.

lanyusea, 2017-03-10:
what's your firmware of the drone?
are you using the latest code in this repository?

lijiangyong01, 2017-03-10:
the firmware version I got from the ios app DjiGo is 1.2.0.90+.
yeah, I'm using the latest code in this repository. did anyone encounter the same problem? thanks.

rohitsantdji, 2017-03-13:
Could you please answer the same questions as those in Issue #77 ?

Additionally, could you please try connecting the UART to the /dev/ttyTHS2 or /dev/ttyTHS1 ports on the TX1 instead of using a USB-Serial adapter?

lijiangyong01, 2017-03-14:
@rohitsantdji I saw the question in Issue#77 carefully, maby it's not the same question?
I haved tried the /dev/ttyTHS2 already, it's the same questions..

natelannan-osu, 2017-03-16:
I am seeing this same issue with a TX1.  Running a Beagle Bone Black I have no issues.
Answers to questions in #77:
1.  baud rate set to 230400 in DJI Assistant and UserConfig.txt
2.  connections are correct (double checked by swapping tx and rx)
3.  api control enabled
4.  UserConfig is correctly populated with ID, Key, and Version
5.  DJI Go is on and powered
6.  pulled latest version of sdk

In addition, I am seeing some erratic serial character ouput to console when running  djiosdk-linux-sample and ptemp is null after unsigned char *ptemp = &(missionACKUnion.droneVersion.ack[0]); in getDroneVersion 

However if I use the old sample CommandlineLinux_old and I change the makefile to use osdk-core and remove any calls to getSDKVersion, setVersion and SDK_VERSION the script works fine.  ptemp is still null, but I am able to automatically activate, gain control using --CA ct 1, and take off using --FC tk 4.  I also don't see any erratic serial output.

Here is an example of erratic output from blocking linux sample:
 ▒>▒▒▒00000000000000000000000000000000▒u Automatic activation successful.
                                                      ▒t~ۈFailed to obtain control.

the old script:
Waiting input
--CA ct 1
Waiting input
STATUS setControlCallback,line 1117: Obtain control running
STATUS setControlCallback,line 1114: Obtained control successfully

any advice would be grand.  Thanks for your time

lijiangyong01, 2017-03-17:
hi natelannan, thank you for your reply. 
I also print the  *ptemp in getDronneVersion, the value in console is NULL. it's the same question.

1. baud rate set to 230400 in DJI Assistant and UserConfig.txt
2. connections are correct (double checked by swapping tx and rx)
3. api control enabled
4. UserConfig is correctly populated with ID, Key, and Version
5. DJI Go is on and powered
6. pulled latest version of sdk

I did all of those steps.The UserConfig in newest onboard sdk ,I cann't find Version config line. Maybe the old onboard sdk need to config the version. 
As you said ,the old sample CommandlineLinux_old removed the any calls to getSDKVersion, setVersion and SDK_VERSION works fine. how can I get the old sample CommanlineLinux_old ? I  need to test the way you suggested.

thanks a lot.

natelannan-osu, 2017-03-17:
CommandlineLinux_old is located in Onboard-SDK/sample/CommandlineLinux_old
it is no longer officially supported, but I just used it as a sanity check to make sure I was communicating correctly with the TX1.  Since the newest version of the SDK doesn't use sdk version data, you'll need to just comment out the lines that reference that.  Also, the file tree changed names, so you'll have to change the makefile to reflect those changes.  Here is specifically what I did:

Onboard-SDK/sample/CommandlineLinux_old/Makefile:
change LIBDIR = ../../lib  to LIBDIR = ../../osdk-core

Onboard-SDK/sample/CommandlineLinux_old/DJIscript/src/cmdCoreAPI.cpp:
comment out the following: 
          if (script->getApi()->getSDKVersion() != versionM100_23)
          {
            cout << "Time: " << bd.timeStamp.time << endl;
            cout << "Nano: " << bd.timeStamp.nanoTime << endl;
            cout << "Sync: " << (int)bd.timeStamp.syncFlag << endl;
          }
          else

Onboard-SDK/sample/CommandlineLinux_old/DJIscript/src/cmdSettings.cpp:
comment out:
script->adata.version = SDK_VERSION;

Onboard-SDK/sample/CommandlineLinux_old/src/main.cpp:
comment out:
api.setVersion(targetVersion);

If I remember correctly, that is all I did (I am away from the drone and TX1 at the moment).  you should be able to then make the old script in CommandlineLinux_old without compile errors.  The resultant script will be in the bin directory.  just run it without any arguments.  It automatically activated for me and then I typed the following to trigger a take off in the simulator:
--CA ct 1  (this takes control of drone if you are in F mode on the controller)
--FC tk 4 (this triggers a take off)
you can do other stuff, but I was mostly interested in doing a sanity check against the latest sample scripts since they weren't working for me using the TX1.  if anyone has any ideas why latest sample scripts aren't working for me, please let me know.

thanks much

rohitsantdji, 2017-04-06:
Hey guys, quick update - we found and resolved a bug in our serial drivers for 64-bit ARM and 32-bit x86 architectures that fixes the erratic prints (and the actual communication itself). We're currently testing this, and we will push a hotfix in the next week.

natelannan-osu, 2017-04-07:
@rohitsantdji Fantastic news.  Thanks so much.

rohitsantdji, 2017-04-20:
Could you guys try to clone the 3.2 branch after the latest commit? In our tests, that did resolve the serial issue on TX1/TX2 devices.

lijiangyong01, 2017-04-21:
@rohitsantdji thank you so much, I'll test it soon.

natelannan-osu, 2017-05-25:
@rohitsantdji Sorry for the delay in responding to this.  I am no longer seeing issues with latest commit.  Thanks much.

adamwathieu, 2017-07-12:
Hey guys,
Could you take a look at my issue here?
https://github.com/dji-sdk/Onboard-SDK/issues/143
Thanks

Jai-GAY, 2018-09-17:
not really sure is it [totally solved](https://github.com/dji-sdk/Onboard-SDK/issues/374#issuecomment-421901219).

closed 2019-05-28 by ChrisChen365