dji-sdk/Onboard-SDK #105: When linux sample "Square" initiated, aircraft immediately turns due north and starts uncommanded climb

tariq-r, 2017-03-13:
Hello

We are using the onboard SDK with the A3 to control an industrial drone.

The Onboard SDK is running on an intel atom single board computer (Minnowboard x86 Ubuntu 14.04 LTS server), we are able to successfully take control of the aircraft using both the DJI OSDK sample application (Linux) and our own application.

Both in the simulator and actual flight we can takeoff and land using the interactive sample

When we try to do the sample "square" live (flying), the aircraft immediately turns due north and starts climbing. We stop the climb at about 15 meters height because the aircraft is on a tether.

The aircraft does not start the square.

In the simulator, the sample works correctly.  That is the aircraft starts the square without turning or climbing first.

I don't know if this is relevant but I reviewed the telemetry and noticed that the altitude (z) indicated about -120m.

We confirmed that the flight controller was in P mode, GPS fix was obtained before takeoff

This behavior was identical on two different A3 flight controllers, two different single board computers.

Can you suggest some corrections?  We need to be able to command precise movement.

Tariq

rohitsantdji, 2017-03-13:
This issue is documented [here](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/69). We will have a fix for you in the next few days.

tariq-r, 2017-03-14:
Thank you, we look forward to it.

rohitsantdji, 2017-03-22:
Closing issue we are tracking this already in OnboardSDK-ROS [Issue 69](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/69)

closed 2017-03-22 by rohitsantdji