dji-sdk/Onboard-SDK #107: Sample code for implementing Virtual RC Protocol?

jrmzelda, 2017-03-13:
Hi, 

I am editing the Linux Interactive Sample and I would like to use the virtual RC protocol for control. Is there an example program that I can modify for my purposes or do I need to build it from scratch?

I have looked at the DJI_VirtualRC.cpp and .h files but I am looking for a sample that I can build on.

*I know that Virtual RC is not suggested, but for my research I need to be able to control the aircraft without the RC turned on. When the RC is on, it jams the routers I am using to send my RPi commands, so I have no option but to find a way to turn the RC off and still control the drones.

Thank you!

rohitsantdji, 2017-03-13:
See the ROS client sample [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk_demo/src/client.cpp#L505).

closed 2017-03-13 by rohitsantdji

jrmzelda, 2017-03-14:
Thank you for the example; however, I also just read this:

"NOTE: Virtual RC has been deprecated and is unavailable for A3 FW > 1.6.0.0 and for N3."

So does this mean that my program will not work?