dji-sdk/Onboard-SDK #108: Stop getting updated PositionData and RadioData (raspberrypi-support branch)

newtoniant, 2017-03-17:
This works on a desktop PC, but not on a Raspberry Pi 2 with the raspberrypi-support branch.

Tracking and printing out location and landing gear:

```
std::cout << getTimestamp() << endl;
std::cout << "LAT:" << std::fixed << std::setprecision(5) << rad2deg(p.latitude) << endl;
std::cout << "LONG:" << std::fixed << std::setprecision(5) << rad2deg(p.longitude) << endl;
std::cout << "ALT:" << std::fixed << std::setprecision(1) << p.altitude << endl << endl;
if (landingGearDown(api){/*do stuff*/}

bool landingGearDown(CoreAPI *api)
{
  bool ret = false;
  RadioData rcData = api->getBroadcastData().rc;
  ret = rcData.gear == -4545;
  std::cout << "landingGearDown [" << rcData.gear << " == -4545?]:" << std::boolalpha << ret << endl;
  return ret;
}

```

I periodically stop getting updated data, and it doesn't come back once it's lost.  On the PC this does not happen.

In this sample output I'm flying, but not getting updated data (moving and toggling landing gear during this period):

```
landingGearDown [-10000 == -4545?]:false
2017-03-17  04:39:44.277588
LAT:22.54238
LONG:113.95925
ALT:11.8

landingGearDown [-10000 == -4545?]:false
2017-03-17  04:39:45.286486
LAT:22.54238
LONG:113.95925
ALT:11.8

landingGearDown [-10000 == -4545?]:false
2017-03-17  04:39:46.294722
LAT:22.54238
LONG:113.95925
ALT:11.8
```

Running on Pi 2, raspberrypi-support branch.

rohitsantdji, 2017-03-21:
@newtoniant : Thanks for the update. We have noticed that USB-Serial adapters throw a bunch of issues on ARM systems (people have reported similar issues on Manifold). Let us know if any issues come up with the ttyAMA0 serial.

jrmzelda, 2017-04-05:
I am having a similar issue. In the DJI Assistant 2 Simulator, I am printing time, lat, long, battery level, and velocity 1 time every second and it works fine before takeoff. Once I takeoff, only the velocity values continue to update. The rest of the data does not update. 

I am using a Raspberry Pi 3 and Virtual RC for control. The Pi is connected to the M100 with the provided GPIO-UART cable.

closed 2019-05-28 by hummelslei