dji-sdk/Onboard-SDK #112: Data Transparent Transmission Usage Scenario 2 - upstream from Mobile Device to OES  for iOS

EonKid, 2017-03-27:
I am building an iOS app in which, i want to send some data to custom hardware which is attached with drone using UART to perform some task by custom hardware.I want to send a flag which will be triggered to perform some action by custom hardware .

Using documentation i have found the command using Data Transparent Transmission
Usage Scenario 2 - upstream from Mobile Device to OES
The following SDK interface can help you understand how to communicate with Onboard SDK Device on different mobile platforms including iOS and Android.
iOS
Please use the following method of DJIFlightController:
- (void)sendDataToOnboardSDKDevice:(NSData *)data withCompletion:(DJICompletionBlock)completion;

The OES can receive the data sent from the Mobile Device by means of a CMD from Autopilot with the CMD Set, ID and Val to be:
CMD Set: 0x02
CMD ID: 0x02
Data Type	Offset(byte)	Size(byte)	Description
CMD Val	0	1~100	User Data
ACK Val	---	---	N/A

Can you give use case example to send data to custom hardware to trigger some action .Like setting a flag to perform an action.
I want packet structure and proper conversion to NSData to send it to custom hardware.Also give me some suggestion how to interact with custom hardware?

amenonDJI, 2017-03-30:
@EonKid [This app](https://github.com/dji-sdk/Mobile-OSDK-iOS-App) sends commands from the iOS app to the OES device (Manifold/x86/STM) and triggers all the sample commands such as Takeoff, Land etc. 

The existing Blocking Linux sample app can be run in "Mobile mode", in which case it receives the commands from the iOS app and triggers the command on the aircraft.  The full pipeline from the iOS app to the OES device and ACK data back to the Mobile app is exposed in our code.

EonKid, 2017-04-03:
@amenonDJI  Hi ,I have reviewed you sample .But it only provide Command Id sending, it does not provide Command set.
Can you shared code for iOS app.For sending both data listed below
    CMD  ID:  0x01
    CMD Set : 0x0C

amenonDJI, 2017-04-03:
@EonKid We don't have code that exactly sends CMD ID and CMD Set. The pipeline and the interfaces that enable you to do this is shown in the app. Please feel free to branch off and make changes to the code.

EonKid, 2017-04-06:
@amenonDJI  Thanks !

closed 2017-04-06 by EonKid