dji-sdk/Onboard-SDK #118: Takeoff fails if LB2 is powered off

taurean1192, 2017-04-26:
I managed to compile the SDK and started up the sample using my Raspi3 as OES. When I tried arming the system with LB2 switched off the motors just spin for about half a second then stop.

When LB2 is switched ON then everything works fine. I thought the job of the Onboard SDK is to make things run independent of the LB2.

rohitsantdji, 2017-05-05:
The LB2 is intended as a safety mechanism to allow users to regain manual control in case the commands made by users through OSDK cause the drone to do unsafe actions. 

If however you want to bypass this safety mechanism, then you can check the `Enable SDK Failsafe Action` button on the top right corner of  the SDK page in DJI Assistant 2. Additionally, set the SDK Failsafe Action to `Hover`.

closed 2017-05-08 by taurean1192

flytrex-vadim, 2017-07-20:
We have exactly the same situation. We tried "Enable SDK Failsafe Action" together with "SDK Failsafe Action: Hover".
The system will complete automatic takeoff to the height of one meter, but then refuses to execute waypoint mission (receives the ACK with the "no error" code but keeps hovering at 1 meter)

When the LB2 signal is lost during the execution of waypoint mission, the state changes to "Hovering" and it stops in place. How can we make it continue the waypoint mission ?