dji-sdk/Onboard-SDK #119: VERT_VEL stopped working in latest firmware

maciekd, 2017-04-26:
Hi,

Im running an A3 using Ubuntu. A3 is on the latest firmware (1.7.x)

Before the latest firmware upgrade I could control the craft using VERT_VEL control mode. Now this does not seem to work - it's as if the flight control commands are ignored completely. VERT_POS works fine.

Is this by design or could it be a bug?

maciekd, 2017-05-02:
Throttle control does not seem to work either. Throttle control is a requirement for our application. :/

maciekd, 2017-05-03:
Turning off the safety max distance in DJI Assistant solved the problem.

maciekd, 2017-05-03:
Sorry, turns out that did not help at all. Still at a loss here.

rohitsantdji, 2017-05-05:
This turned out to be a bug on the OSDK. We just pushed a hotfix, could you please try it out?

maciekd, 2017-05-06:
It works! Thanks for the help.

closed 2017-05-06 by maciekd