dji-sdk/Onboard-SDK #120: 1.0.0.80 firmware issues

tudorDME, 2017-04-27:
We have had an app that is loosely based on the Onboard SDK 3.1 sample/Linux/Blocking reference app working for the past couple of months. The problem is that all of our users that have been tempted to upgrade to 1.0.0.80 or who preferred to purchase the M600 Pro have experienced communications issues. The M600 users at least have the option of downgrading to continue being supported. Despite lots of testing, we still can't get the M600 Pro to work.

Here are our findings:
In 3.1, the M600 Pro gets hung up in how it reports its version_ID since that isn't used in the 3.2 anymore. Fine. After updating our app to 3.2.2, which we thought might fix the issues, it still isn't working. The aircraft is properly identified, gets authorized and obtains control just fine but we get a long stream of "ERROR recvReqData,line 248: Unknown mission code 0x30" messages. Upgrading to 3.2.2 fixed one issue and introduced another. The updated 3.2.2 app works fine with the M600 with the firmware downgraded to 1.0.0.43.

Here is the output of the blocking sample on **M600** fw 1.0.0.43 built with SDK 3.2.2 :
`User Configuration read successfully.

These are your User_Config settings.

Serial port = /dev/ttyMFD1
Baudrate = 115200
App ID: xxx
App Key: xxx

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make$

STATUS init,line 56: Attempting to open device /dev/ttyMFD1 with baudrate 11520$
STATUS init,line 65: ...Serial started successfully.
STATUS parseDroneVersionInfo,line 430: Device Serial No. = 0670129101
STATUS parseDroneVersionInfo,line 433: Hardware = PM820V3
STATUS parseDroneVersionInfo,line 435: Firmware = 3.1.100.0
STATUS parseDroneVersionInfo,line 439: Version CRC = 0x2CBD2A18

Attempting activation..
Automatic activation successful.
Obtained control successfully.`

Here is the output of the blocking sample on **M600 Pro** fw 1.0.0.80 built with SDK 3.2.2 :
`User Configuration read successfully.

These are your User_Config settings.

Serial port = /dev/ttyMFD1
Baudrate = 115200
App ID: xxx
App Key: xxx

Does everything look correct? If not, navigate to Linux/UserConfig.txt and make changes.

STATUS init,line 56: Attempting to open device /dev/ttyMFD1 with baudrate 115200...
STATUS init,line 65: ...Serial started successfully.
ERROR recvReqData,line 248: Unknown mission code 0x30
ERROR recvReqData,line 248: Unknown mission code 0x30
ERROR recvReqData,line 248: Unknown mission code 0x30
ERROR recvReqData,line 248: Unknown mission code 0x30
ERROR recvReqData,line 248: Unknown mission code 0x30
ERROR parseDroneVersionInfo,line 269: Drone version was not obtained. Please restart the
program or call getDroneVersion

Attempting activation..
STATUS parseDroneVersionInfo,line 430: Device Serial No. = 0670130329
STATUS parseDroneVersionInfo,line 433: Hardware = PM820V3PRO
STATUS ,line 435: Firmware = 3.2.100.0

Automatic activation successful.
Obtained control successfully.`

What is causing this "Unknown mission" code error in the 3.2.2 blocking sample on a M600 Pro with 1.0.0.80 on it?

Are the sample apps maintained and tested against 1.0.0.80?

Will there be an update soon to fix this?

tudorDME, 2017-05-10:
Everything works now with the new firmware that was released 5/2/2017 works (v1.0.1.20).

rohitsantdji, 2017-05-23:
That's good to know!

closed 2017-05-23 by rohitsantdji