dji-sdk/Onboard-SDK #122: Odroid XU4 +ROS + Matrice 100_Error 449

AWKIONG, 2017-05-01:
Hey guys, need the community advice on this matter. Been cracking my skull over this.

Trying to link my Odroid XU4 (Ubuntu 14.04 + Onboard SDK ROS 3.2) with my Matrice 100. Received the following message:
[image]

I have read through issues #77, #104 and config my setup according:
- Latest firmware for DJI Go app, Remote control,  Matrice 100, Onboard-SDJ-ROS 3.2

- All connections are made; 
    DJI GO app connected to Remote Controller
    Matrice 100 connected to Odroid GPIO with careful emphasis on Tx and Rx matching
    Matrice 100 connected to Laptop via USB
    Running DJI Assistant 2, with API checked
    Running SSH connection to my Odroid (with internet connection)

- Configured the sdk_manifold.launch file:
    Updated App ID and Key
    Serial set to `ttySAC0`
    Baud rate unchanged at 230400

- Toggled the F mode around, kept it and ran it in F mode

My initial thought is that I got the serial_name wrong. So I experimented with all the available ttySAC* available from `dmseg | grep tty*`

Any recommendations or suggestions?

rohitsantdji, 2017-05-01:
Did you add your user to the `dialout` group using `sudo usermod -a -G dialout $USER`?

AWKIONG, 2017-05-02:
@rohitsantdji, I have checked the groups rights.
[image]

Some more photos for troubleshooting. 
[image]

To verify the wiring of Odroid GPIO
[image]

I noticed for the new firmware of the Remote Controller, we do not have the option to turn off the IOC. The settings only have the multiple flight modes. I have also tried toggling the F mode around, and leaving it on F mode before `roslaunch`

=|

chuong98, 2018-01-05:
The odroid xu4 GPIO use 1.8V level, while most of the other external uses 3.3V. So you need a level shift converter to connect them.

closed 2019-05-28 by hummelslei