dji-sdk/Onboard-SDK #123: Aircraft did not respond to moveWithVelocity command

tariq-r, 2017-05-04:
Hello

We have been experimenting with the moveWithVelocity from LinuxFlight.cpp.   

We have varied the velocity across and range and the timeout across a range but the aircraft does not appear to respond to the command.

Flight controller: DJI A3
Code attached, console log output attached

[console-log.txt](https://github.com/dji-sdk/Onboard-SDK/files/977435/console-log.txt)
[code-snippet.txt](https://github.com/dji-sdk/Onboard-SDK/files/977439/code-snippet.txt)

tariq-r, 2017-05-04:
We did some experimentation and found that 
01010011
0x53  (Body Frame, yaw angle, vert pos) gives the desired behavior though this is not what we expected the meaning of body frame of reference to be.

rohitsantdji, 2017-05-05:
We just pushed a hotfix, can you confirm that solves your issue?

tariq-r, 2017-05-05:
Using 0x53 fixed the issue as a workaround using the SDK before the hotfix, we will pull the hotfix and try it in a couple of days.  Thanks very much.

closed 2017-05-23 by rohitsantdji