dji-sdk/Onboard-SDK #125: OnBoard SDK can't connect to Matrice 100

lfway, 2017-05-11:
Hi! I have built PureQT example for Raspberry Pi 3.
I enabled API control in DJI Assistant2 for Matrice.
I connectted M100 to RPi using UART-CAN2 cable.
I registered as DJI developer and I getted app key and app id.
I copied getted app key and app id to PureQT.
I opened ttyAMA0 port
BUT
When I click "Version" program is doing nothing.
When I click "Activate" I see "ERROR activate,line 451: Please call getDroneVersion first."
[image]
I tryed with different baud rates.
Help me please!

kkoppul2, 2017-07-07:
I am having the same issue with getDroneVersion either not being called or not returning anything to the version data struct. I am using the Manifold OES, but I believe that this issue may have common cause for both of us. Have you had any luck fixing this issue?

adamwathieu, 2017-07-11:
I have a similar issue, any help would be great:
https://github.com/dji-sdk/Onboard-SDK/issues/143

OmkarSavant, 2017-07-17:
+1 Also having this issue

justwillim, 2017-07-27:
hi all, If you are using ubuntu,  try to use this sample under sudo, make sure you have already added usergroup dialout.

justwillim, 2017-07-27:
if all these movement you already tried, please ensure your hardware connection is right, use cat /dev/YOURTTY_DEVICE to see if the FC is sending out datas.

Jai-GAY, 2018-09-17:
It seems like many [experienced this issue](https://github.com/dji-sdk/Onboard-SDK/issues/374#issuecomment-421901219).

closed 2019-05-28 by hummelslei