dji-sdk/Onboard-SDK #132: missing include for std::end

Trass3r, 2017-06-16:
```diff
diff --git a/osdk-core/platform/linux/src/linux_serial_device.cpp b/osdk-core/platform/linux/src/linux_serial_device.cpp
index 755ca88..9c92c74 100644
--- a/osdk-core/platform/linux/src/linux_serial_device.cpp
+++ b/osdk-core/platform/linux/src/linux_serial_device.cpp
@@ -14,6 +14,8 @@
 
 #include "linux_serial_device.hpp"
 #include <algorithm>
+#include <iterator>
+
 using namespace DJI::OSDK;
 
 /*! Implementing inherited functions from abstract class DJI_HardDriver */
```

rohitsantdji, 2017-06-16:
Could you provide your platform and what version of gcc you are using?

Trass3r, 2017-06-19:
Ubuntu 17.04 with gcc 6.3

rohitsantdji, 2017-06-20:
Thanks! We don't yet test on Ubuntu 17.04. I'll push a fix soon.

rohitsantdji, 2017-06-23:
Addressed in e6c293341fb5461e8b364a584ca2bae8855cb5d5

closed 2017-06-23 by rohitsantdji