dji-sdk/Onboard-SDK #133: Matrice 100 Support

GitPlz, 2017-06-20:
The documentation at https://developer.dji.com/onboard-sdk/documentation/introduction/homepage.html has a note stating "Note. OSDK 3.3 does not support Matrice 100". 

What does that statement mean exactly?
Will version 3.3 no longer work with the Matrice 100, or are there some subset of feature that won't work?
Will any future versions support the Matrice 100?
Can I continue to use version 3.2 indefinitely with the Matrice 100, even if I update the firmware?

rohitsantdji, 2017-06-21:
You can continue to use version 3.2 indefinitely on the Matrice 100 even if you update the firmware. The main challenge with support for the Matrice 100 is a lack of firmware-side support for new OSDK features being introduced for the other platforms - and this is a hardware limitation that cannot be overcome easily.

Currently, the 3.3 release will exit when connected to an M100, there is no partial compatibility.

That said, we are actively working to bring compatibility (for the feature sets that do exist on the M100) to OSDK 3.3, and a future release will support the M100 to the fullest extent possible.

closed 2017-06-21 by rohitsantdji

GitPlz, 2017-06-23:
That isn't what the developer team emailed out. They state that you can't update your firmware:

If you are coming from OSDK 3.2, please note that you will have to upgrade your firmware to take advantage of OSDK 3.3. Conversely, the newer firmware will not work with the older OSDK version.
 
Regards,
 
The DJI Dev Team

rohitsantdji, 2017-06-23:
The version map on the doc page [here](developer.dji.com/onboard-sdk/documentation/appendix/versioning.html) is the source of truth for compatibility. The "newer firmware" mentioned in the dev email refers to firmware that has been released around the time of OSDK-3.3 - M100 firmware was last updated in 2016, so even the newest M100 firmware falls into the "older firmware" bracket. Sorry if that wasn't well articulated in the email!