dji-sdk/Onboard-SDK #135: OSDK 3.3 - Telemetry sample crashes with "Bus error"

hhackbarth, 2017-06-26:
Testing the telemetry sample on a RasPi Zero.

Everything seems to work fine until the loop starts. When a value is retrieved with
vehicle->subscribe->getValue<TOPIC...>()
the program crashes after reporting "Bus error" to the console.

Clean termination with removePackage is not carried out. If the program is started a second time, after initialization it immediately reports "ERROR/1 @ extractOnePackage, L364: Package does not have a valid DataBuffer".

I tried to reduce the amount of data by only subsribing to package 0 and 1 with reduced frequency. This did not help. Also reducing the output frequencies in the SDK configuration did not help.

rohitsantdji, 2017-06-26:
The general opinion on various forums ([example 1](https://github.com/dart-lang/sdk/issues/22384), [example 2](https://wiki.linuxaudio.org/wiki/raspberrypi)) is that variants of the Raspberry Pi do not like unaligned memory access (packed structs), which are used quite frequently in the DJI OSDK. 

You could try going in to dji_telemetry.hpp and commenting out the #pragma packing, though we have not tested the SDK using this configuration so it is unexplored territory.

hhackbarth, 2017-06-26:
Thank you for your quick comment!

I tested it with commenting out the #pragma packing in dji_telemetry.hpp:
It does'nt crash anymore but now it shows weird figures for the latitude, longitude and altitude.

Looks like the telemetry cannot be used on RasPi (tested with Pi Zero and Pi 2 B).

closed 2017-06-26 by hhackbarth